mletenay/home-assistant-goodwe-inverter #214: State class total_increasing, but its state is not strictly increasing

TomEkk33, 2023-10-18:
Hello 

Today I noticed this kind of warning "state class total_increasing, but its state is not strictly increasing"
HACS GW v0.9.9.11

Home Assistant 2023.10.1 
Supervisor 2023.10.0 
Operating System 10.4

I'm not sure if it is important or not. Thanks for checking.

```
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:298
Integration: Sensor (documentation, issues)
First occurred: 11:10:10 (6 occurrences)
Last logged: 11:10:10

Entity sensor.today_s_pv_generation from integration goodwe has state class total_increasing, but its state is not strictly increasing. Triggered by state 4.8 (5.1) with last_updated set to 2023-10-18T09:08:03.098242+00:00. Please report it to the custom integration author.
Entity sensor.total_load from integration goodwe has state class total_increasing, but its state is not strictly increasing. Triggered by state 7592.6 (7592.7) with last_updated set to 2023-10-18T09:08:03.098623+00:00. Please report it to the custom integration author.
Entity sensor.today_load from integration goodwe has state class total_increasing, but its state is not strictly increasing. Triggered by state 7.8 (7.9) with last_updated set to 2023-10-18T09:08:03.101790+00:00. Please report it to the custom integration author.
Entity sensor.total_battery_charge from integration goodwe has state class total_increasing, but its state is not strictly increasing. Triggered by state 4995.2 (4995.5) with last_updated set to 2023-10-18T09:08:03.102029+00:00. Please report it to the custom integration author.
Entity sensor.today_battery_charge from integration goodwe has state class total_increasing, but its state is not strictly increasing. Triggered by state 5.3 (5.6) with last_updated set to 2023-10-18T09:08:03.102211+00:00. Please report it to the custom integration author.
```

beilm, 2023-12-29:
I have had a similar error message for some time:

2023-12-29 21:05:10.394 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.today_s_pv_generation from integration goodwe has state class total_increasing, but its state is negative. Triggered by state -459.8 with last_updated set to 2023-12-29T19:59:59.999999+00:00. Please create a bug report at https://github.com/mletenay/home-assistant-goodwe-inverter/issues

Is there any way to help analyze the error?

TomEkk33, 2024-07-24:
No longer exists

closed 2024-07-24 by TomEkk33