mletenay/home-assistant-goodwe-inverter #217: 3 MPPT Inverters, wrong assign PVx entities

AT622CZ, 2023-10-26:
Hi,
GW25K-ET or GW30K-ET has SIX PV inputs and 3 MPPTs. Each input has separate sets of registers. Voltage of PV1 and PV2 (PV3, PV4 and PV5, PV6)  are same, but Currents are measured individually.  I have two strings connected to the first and second MPPTs, inputs 1 and 3.  Correct values I need read from PV1, PV3. PV5 is unconnected now.

[image]

Best regards
Michal Patek

mletenay, 2023-10-26:
Hmm, this seems odd.
The known map of modbus registers have 4 sets of voltage/current/power for potential 4 MPPTs.
See https://github.com/marcelblijleven/goodwe/blob/90909e06c6fa2088379503aca10e27a12c397e87/goodwe/et.py#L22
If these report wrong valules and there are different/additional registers for ETT inverters, well, I do not know them.

mletenay, 2023-10-26:
For reference - https://github.com/home-assistant/core/issues/95703

AT622CZ, 2023-10-27:
> Hmm, this seems odd. The known map of modbus registers have 4 sets of voltage/current/power for potential 4 MPPTs. See [...]

Throw modus, for MPPT 1 and 2 we can use this set of register (PV1-PV4). BUT! Only Voltage and current (seams)!. Correct Voltages and Currents I can read from PV1 and PV3 regs, MPPT 1 power is in PV1, MPPT2 is PV2.
(on pictures: today is rainy, dark, MPPT 3 is not connected now).

Adresses of PV5 and PV6 I don't know :-( GW support is silent.

[image]

[image]

mletenay, 2023-10-27:
Can you monitor all 4 those V/A/W values via SolarGo mobile app ?

AT622CZ, 2023-10-28:
SolarGo shoes V/A, all 6. But as I wrote PV1-PV4 are same registers as on 10kW GW. MPPT1 is PV1, PV2, MPPT2 is PV3-PV4. MPPT3 we don't know adresses. Power for MPPT1 is PV1, MPPT2 is PV2, MPPT3 is PV3

[image]

mletenay, 2023-12-03:
See https://github.com/marcelblijleven/goodwe/issues/62
There was some progress with ETT inverters, once I'll release the library I'll adjust the HA integration

sergioafeo, 2023-12-06:
Thanks @mletenay. I have a similar issue with  GW15ET. It has 2 MPPTs and 4 inputs. Strings are connected to inputs PV1 and PV3. Both pyhton lib and HA integration show up to PV2 only.

mletenay, 2023-12-12:
The v0.9.9.13 was just release and it contains proper support for ETT, incl. 2nd battery, pv1-6 and mppt sensors.

closed 2023-12-12 by mletenay