mletenay/home-assistant-goodwe-inverter #218: Fast Charging not working on ES models

drydenmike, 2023-10-27:
Hey. Just to let you know I appreciate the work you're doing here.
Having said this, I have a ES model and fast charging is not working with any of the parameters mentioned hereÇ https://github.com/mletenay/home-assistant-goodwe-inverter/issues/150

Do you think this might be different depending on model/software version?

mletenay, 2023-11-07:
Yes, it is possible it is firmware/model specific and since I do not have direct access to different inverters I cannot test/fix that (if possible at all).

drydenmike, 2023-11-10:
Is there something that I can do at my end to extract this information?

traveler6666, 2024-04-16:
Hi,

I really love this featrue. However, shortly after I set up the fast charging service, I can't set the SOC anymore. I think there was an update.
I am getting an unpexted error message and the protocoll says: goodwe.exceptions.RequestRejectedException: ILLEGAL DATA VALUE if the value is e.g. 81 or input_number.fast_charging_soc.

The code is:

service: goodwe.set_parameter
data:
device_id: b9cb944ba3a1382249238c4a2523b698
parameter: fast_charging_soc
value: states("input_number.fast_charging_soc")

Is there an error in the code or a bug in the integration, as it worked before?

drydenmike, 2024-05-23:
This is working in the latest versions.

closed 2024-05-23 by drydenmike