mletenay/home-assistant-goodwe-inverter #219: 5048ES loses controls on version v0.9.9.12

tinuva, 2023-10-28:
I had to roll back to version v0.9.9.11

Inverter: 5048ES with firmware 2424H

On the next version I lose the following controls:

* Inverter operation mode
* Eco mode power
* depth of discharge

Strangely, eco mode soc, was still available.

Sascha883, 2023-11-01:
Hello, since the update to v0.9.9.12 the same thing here too.

Back to v0.9.9.11 without success

Inverter update to 060608

GW6000ES20 Firmware: 04048-06-S06 / 02020-08-S01

Home Assistant 2023.10.5
Supervisor 2023.10.1
Operating System 11.1
Frontend 20231005.0 - latest

davidkm123, 2023-11-07:
I have the same problem as @tinuva.

Changing depth_of_discharge_on_grid in HA is not transferred to the inverter and when the value in solar go changes, it is not reflected in HA.

Changing the mode in the HA will change the inverter but will not start charging.

When I set the timer in solar go, it starts charging, but I still see the general mode in the HA

How can I go back to a previous version?

Thanks David

mletenay, 2023-11-07:
I cannot find out anything wrong in the code.
Just to be sure, you say the 0.9.9.11 does work properly and in 0.9.9.12 some settings are gone ?

tinuva, 2023-11-08:
Hmm. So I am now on home assistant 2023.11.1 and loaded up 0.9.9.12 again, and now those inputs are still available. I even went as far as enabling debug logging for the component just in case. Dont know how to explain this, but it definitely was an issue the first time I upgraded.

davidkm123, 2023-11-08:
> Nemohu v kódu najít nic špatného. Pro jistotu říkáte, že 0.9.9.11 funguje správně a v 0.9.9.12 jsou některá nastavení pryč?

> I cannot find out anything wrong in the code. Just to be sure, you say the 0.9.9.11 does work properly and in 0.9.9.12 some settings are gone ?

I have run further tests and cannot now 100% confirm that the problem is with the plugin. My PV contractor has a SunBerry (Rasberry) plugged into the cabinet and it is possible that it overrides the goodwe setup.

I have verified that it overrides number.depth_of_discharge_on_grid and when I set eco_charge_mode and reload the integration, it changes to Eco mode in HA mode.

davidkm123, 2023-11-10:
Hey. Hey, 

I'm detecting this strange behavior.

I turn on eco_charge and after a while, within an hour at most, the charging turns off and when I reload the Goodwe device, I see eco mode in HA and the values number.eco_mode_power and number.eco_mode_soc are both at 0.

What is wrong? 

David

rans100, 2023-11-12:
I have the same problem, when I upgrade to HA 2023.11(.02) it seems like I lose control over my inverter (5048ES), I was already on Goodwe integration version 0.9.9.12. 
I lose control over all the functions with input numbers (deapth of discharge, export limit,...). See screenshot attached, no slider for input D.O.D. visible or available.
Rolling back to 0.9.9.11 doesn't help, reinstalling 0.9.9.12 neither. 
Had to roll back to HA 2023.10.05

[image]

rans100, 2023-12-11:
I don't know what caused this problem but I just tried to update to HA 2023.12.(01) and everything seems fine now.
I do have all the controls back with Goodwe integration 0.9.9.12.
Strange... But for me case closed!

closed 2023-12-21 by mletenay