mletenay/home-assistant-goodwe-inverter #220: Household power consumption - smart meter, no homekit

mkerollos, 2023-11-04:
Hi,
Is it at all possible to get consumption values from the inverter? I could not see anything that resembled consumption.
I have a [GoodWe smart meter ](https://en.goodwe.com/smart-energy-meter) but do not have a homekit. The consumption values show up in SEMS (and I can retrieve them using the SEMS portal, but I would prefer local integration).

mletenay, 2023-12-21:
All "Meter" prefixed sensors are coming from the smart meter, so those are the ones measuring actual or cumulative consumption/production.

closed 2023-12-21 by mletenay

mkerollos, 2024-02-12:
@mletenay there are no "meter" prefixed sensors for me. I have a GW10K-MS-30 inverter with a smart meter. On the SEMS app, I get both consumption and generation. 
[image]
[image]

Perhaps this issue should be against the python SDK, as these are the only values I get out of that:
```
timestamp:               Timestamp = 2024-02-12 15:45:06
vpv1:            PV1 Voltage = 265.4 V
ipv1:            PV1 Current = 10.4 A
ppv1:            PV1 Power = 2760 W
vpv2:            PV2 Voltage = 329.1 V
ipv2:            PV2 Current = 10.4 A
ppv2:            PV2 Power = 3423 W
vpv3:            PV3 Voltage = 296.5 V
ipv3:            PV3 Current = 10.3 A
ppv3:            PV3 Power = 3054 W
vline1:                  On-grid L1-L2 Voltage = -0.1 V
vgrid1:                  On-grid L1 Voltage = 242.2 V
igrid1:                  On-grid L1 Current = 37.1 A
fgrid1:                  On-grid L1 Frequency = 50.01 Hz
pgrid1:                  On-grid L1 Power = 8986 W
ppv:             PV Power = 8886 W
work_mode:               Work Mode code = 1
work_mode_label:                 Work Mode = Normal
error_codes:             Error Codes = 0
warning_code:            Warning code = 0
apparent_power:                  Apparent Power = 8915 VA
reactive_power:                  Reactive Power = -720 var
temperature:             Inverter Temperature = 53.2 C
e_day:           Today's PV Generation = 63.1 kWh
e_total:                 Total PV Generation = 6399.4 kWh
h_total:                 Hours Total = 1434 h
safety_country:                  Safety Country code = 9
safety_country_label:            Safety Country = Australia A
funbit:                  FunBit = -1
vbus:            Bus Voltage = 377.6 V
vnbus:           NBus Voltage = -0.1 V
derating_mode:           Derating Mode code = -1
derating_mode_label:             Derating Mode =
```

alduffield, 2024-10-07:
Did you ever get to the bottom of this? I’m also scrabbling around like a deer in headlights trying to find a way to expose the values that are obviously included in the data package going to SEMS but apparently not exposed locally.

mkerollos, 2024-10-07:
@alduffield unfortunately not, I fell back to using the values out of the sems integration.
I tried poking around the modbus stream to see if I could find anything that sort of lined up but it felt futile.

alduffield, 2024-10-07:
That’s good to know, I’m essentially on the same journey as you, my next step was to pickup a RS485 to Ethernet interface and start poking around, you may have saved me the hassle. I’ll probably just settle for a Shelly or similar and handle integration in HA instead of chasing the elegant solution.