mletenay/home-assistant-goodwe-inverter #223: Eco charge mode not working at night

yutani42, 2023-11-22:
First of all: thank You for this great integration!

I have a seemingly weird problem: at day times eco mode charge works fine, but not at night (when prices are low!).
It once worked, when the battery had some soc left and was (slowly) discharging to home...

Recently I enabled battery charging limit sensor and noticed the following:
[image]

Maybe this is the problem? Is there a way to override the charging limit (e.g. via the set-parameter service)?

Thanks in advance.

yutani42, 2023-11-23:
This night it worked again:
[image]
(charged from 3-5am)
I need control over that charge limit..
(One detail I missed saying before: There is an Entity called FEMS (battery software) which interferes..)

mletenay, 2023-12-21:
The battery charge limits should by controlled by the BMS, not the inverter. 
I do not recommend messing around with set-parameter service unless you know exactly what and why are you doing.

closed 2023-12-21 by mletenay