mletenay/home-assistant-goodwe-inverter #226: Error when switching to off grid mode

leonw1090, 2023-11-29:
Hi all!

I have the following error when switching to off grid mode. I want to force my goodwe to go off grid to protect the house from power spikes.

Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:230
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: November 24, 2023 at 4:25:57 PM (2 occurrences)
Last logged: 7:29:27 PM

[140089903690944] ('No valid response received even after 5 retries', 1)
[140089710700352] ('No valid response received even after 5 retries', 1)
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/goodwe/inverter.py", line 132, in _read_from_socket
    result = await command.execute(self.host, self.timeout, self.retries)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/goodwe/protocol.py", line 109, in execute
    await response_future
goodwe.exceptions.MaxRetriesException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 230, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 2035, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2072, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 876, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 106, in async_select_option
    await entity.async_select_option(option)
  File "/config/custom_components/goodwe/select.py", line 136, in async_select_option
    await self._inverter.set_operation_mode(
  File "/usr/local/lib/python3.11/site-packages/goodwe/es.py", line 305, in set_operation_mode
    await self._set_offgrid_mode()
  File "/usr/local/lib/python3.11/site-packages/goodwe/es.py", line 364, in _set_offgrid_mode
    await self._set_relay_control(3)
  File "/usr/local/lib/python3.11/site-packages/goodwe/es.py", line 414, in _set_relay_control
    await self._read_from_socket(Aa55ProtocolCommand("03270200" + "{:02x}".format(param), "03B7"))
  File "/usr/local/lib/python3.11/site-packages/goodwe/inverter.py", line 137, in _read_from_socket
    raise RequestFailedException(f'No valid response received even after {self.retries} retries',
goodwe.exceptions.RequestFailedException: ('No valid response received even after 5 retries', 1)

tinuva, 2023-11-30:
I get the same error on the 5048ES.

The service is able to successfully set off-grid mode on the inverter, but because of this error, the integration still think the inverter is in the previous mode (eg. general or eco mode). So if you want to go back to the previous mode, you have to set another mode first. Eg. if you were in general and offgrid mode, you first have to set it to eco, then you can set it back to general to leave offgrid mode.

mletenay, 2023-12-21:
Is it just problems of reliability of the commands ? (In that case increasing retry-count might help).
Or is it problem that the call to _set_relay_control always fails with your firmware version ?

leonw1090, 2023-12-21:
To the best of my fault finding I belive it is the parameter "set relay control" it always fails. And the HA gives the timeout were it outputs that after 10 tries the parameter has failed.Goodwe has updated my firmware last week to 2525J and the issue is still there 🙂Can you maby disable that relay control for the ES and then I can test for you?

robotz0mbie, 2024-02-16:
I get the same problem on 5048ES, as @tinuva mentioned change to eco first then normal works.