mletenay/home-assistant-goodwe-inverter #229: ET series - ECO charge mode writes Battery Rated Power to 0%

frpaar, 2023-12-10:
Hello,

I have a GoodWe ET series inverter together with LX F16-H battery. 

I would like to control the charging based on electricity price. For charging, I planned to set the inverter into "ECO charge mode" together with ECO mode power and ECO mode SOC. Rest of the time, the inverter would be in General mode.

Switching between operation modes works, but as soon as I switch to ECO charge mode, the Battery Rated Power is written to 0% (regardless of the ECO mode power set from HA or value inserted in SEMS portal beforehand).
For example, in the following figure I had configured the Battery Rated Power to 100% in SEMS portal beforehand. After setting inverter to ECO charge mode from HA, the value is overwritten to 0%.
[image]

[image]

Also, I have noticed that I do not have option for ECO mode SOC in SEMS portal that some have. Could this be the reason for the issue?

Inverter details:
Model: GW10KN-ET
Firmware version: 090925 (from SolarGo app)
Device information from SolarGo app:
- Control version: 09.163
- Transmittal version: 25.250
- Battery version: 02.01

Has anyone experienced something like this or could point out what's wrong in my case?
Thank you in advance!

lampra1, 2023-12-23:
I also have a GoodWe ET series inverter together with LX F16-H battery.
To my understanding, changes from HA, at least the operation modes, are not reflected in the sems app (through internet connection). I didn't test with solargo with direct connection.
For me, everything works through HA using automations, including eco-mode power and eco-mode soc. I am experiencing some minor issues with eco charge mode (BMS stops charging below 100% SOC when battery temp is higher than 25oC). Keep in mind that eco charge and eco discharge @ 0% power both emulate General mode. So you do not need to change the mode if you only need charge and general. Just set the power to 0%  when you need General mode

jebeke65, 2024-01-26:
I can easily switch operation modes (General vs ECO charge mode) on a GW5048D-ES. 
The charging current I modify by setting the Eco mode power to e.g. 30%, which works perfectly fine

bulkan, 2024-05-29:
This is probably the wrong place to ask but I can't seem to find information on what `Battery Rated Power` setting means in the Eco settings. 

What does it mean if i set to `Discharging 100%` or `Charging 80%` ?

lampra1, 2024-05-29:
@bulkan 
Eco Discharging 100%: means 100% of the inverter power. 
For a 5kw inverter, this means discharging of the battery @5kw 
Eco Charging 80%: means 80% of the inverter power. 
For a 5kw inverter, this means charging of the battery @4kw from the grid