mletenay/home-assistant-goodwe-inverter #230: Getting started

hemiceni, 2023-12-14:
Im new to my goodwe inverter and havent been able to learn all the ins ans outs yet. 

I've read through the readme, and all the comments here. I've googled for days but I'm really having trouble finding out what different modes do and how to be affective in making the inverter do what i want.

Is there somewhere i can find a list or documentation of which settings have which effect in thisintegration?

Thank you

abaksa1, 2023-12-16:
I was in your shoes couple month ago.
My learning path was mostly by experimenting with controls and observing how these effect power flows.
What I found helpful is to have all relevant controls (modes, power settings, DoD settings) AND power flow diagram in one dashboard in Home assistant.
After you learned what controls are having effect on what flows, you can start building automations.

hemiceni, 2023-12-17:
Thanks, i think i will do something like that once i get the intergration working properly

elRadix, 2023-12-17:
You can use this card to get more inside about the overall goodwe sensor
[image]

Card available here:
https://github.com/slipx06/sunsynk-power-flow-card

here you find the goodwe sensor to include in your card:
https://slipx06.github.io/sunsynk-power-flow-card/examples/goodwe.html

mletenay, 2023-12-21:
If something from this ticket can be useful to beginners, please copy it to discussions section and follow there.
I'm closing this issue, since it's not an issue but a question.

closed 2023-12-21 by mletenay