mletenay/home-assistant-goodwe-inverter #231: setting working mode doesn't take effect goodwe EH interter

hemiceni, 2023-12-18:
Hi, hoping for a little guidance/

Using the integration I can view all sensors, and I can also successfully increase and decrease export limit.

But - whilst any working mode I set via home assistant will be reflected in solarGo app, the interter doesnt actually change how it operates.

If I use sems app to change working modes, I see the expected results.

From what I have been able to find in documentation here this may be a firmware issue - can you confirm? and if so what firmware version do i need?

GW6000N EH inverter
Inverter firmware 040418 (as showing in solarGo app)

Thank you

hemiceni, 2023-12-19:
To update, i have now noticed i can successfully set general mode from Home assistant, but still have the issue setting eco modes. I would really appreciate any help with this one, if theres any additional info i need to supply please let me know.

mletenay, 2023-12-21:
Which modes do and which do not work ?
Are there failure responses in the logs ?

hemiceni, 2023-12-24:
Setting both Depth of Discharge, and Export Limit works perfectly.

Setting any/all of the work modes in home assistant will update the "working mode" status on the SolarGo home page.
However only these 3 modes appear to be actually working when set from home assistant - (I can see the change in power flow)

General mode
Backup mode
Off-grid mode (At least I think off-grid working. although I still see active power  between -100w and 100w so not entirely off grid?)

Eco, Eco Discharge, Eco charge, do not do anything when set in home assistant.
If I use the sems or solargo apps to set eco mode, then I notice the power flow change accordingly in home assistant.

In my testing i did see this error pop up once:
Steps I took before error:
-Set eco discharge mode using Sems App
-Then set general mode via home assistant.

```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:230
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 00:58:44 (1 occurrences)
Last logged: 00:58:44

[547062159296] ILLEGAL DATA ADDRESS
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 230, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 2035, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2072, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 876, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 106, in async_select_option
    await entity.async_select_option(option)
  File "/config/custom_components/goodwe/select.py", line 136, in async_select_option
    await self._inverter.set_operation_mode(
  File "/usr/local/lib/python3.11/site-packages/goodwe/et.py", line 446, in set_operation_mode
    await self.write_setting('backup_supply', 1)
  File "/usr/local/lib/python3.11/site-packages/goodwe/et.py", line 394, in write_setting
    await self._read_from_socket(ModbusWriteCommand(self.comm_addr, setting.offset, value))
  File "/usr/local/lib/python3.11/site-packages/goodwe/inverter.py", line 132, in _read_from_socket
    result = await command.execute(self.host, self.timeout, self.retries)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/goodwe/protocol.py", line 109, in execute
    await response_future
  File "/usr/local/lib/python3.11/site-packages/goodwe/protocol.py", line 48, in datagram_received
    if self.command.validator(data):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/goodwe/protocol.py", line 236, in <lambda>
    lambda x: validate_modbus_response(x, cmd, offset, value),
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/goodwe/modbus.py", line 147, in validate_modbus_response
    raise RequestRejectedException(failure_code)
goodwe.exceptions.RequestRejectedException: ILLEGAL DATA ADDRESS
```

The only other thing I can find in my logs are these, are they relevant?

```
	Line   5672: 2023-11-30 10:16:45.512 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.rssi (<class 'custom_components.goodwe.sensor.InverterSensor'>) took 0.790 seconds. Please create a bug report at https://github.com/mletenay/home-assistant-goodwe-inverter/issues
	Line   6647: 2023-12-02 00:40:20.520 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.meter_power_factor_l1 (<class 'custom_components.goodwe.sensor.InverterSensor'>) took 0.792 seconds. Please create a bug report at https://github.com/mletenay/home-assistant-goodwe-inverter/issues
	Line   6843: 2023-12-02 11:20:49.456 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.meter_total_energy_export (<class 'custom_components.goodwe.sensor.InverterSensor'>) took 0.730 seconds. Please create a bug report at https://github.com/mletenay/home-assistant-goodwe-inverter/issues
	Line   9059: 2023-12-05 13:36:57.723 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.meter_active_power_l1 (<class 'custom_components.goodwe.sensor.InverterSensor'>) took 0.996 seconds. Please create a bug report at https://github.com/mletenay/home-assistant-goodwe-inverter/issues
	Line  17201: 2023-12-11 10:58:47.789 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.battery_min_cell_temperature (<class 'custom_components.goodwe.sensor.InverterSensor'>) took 1.055 seconds. Please create a bug report at https://github.com/mletenay/home-assistant-goodwe-inverter/issues
[... cut ...]
	Line  31790: 2023-12-18 00:08:46.828 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.battery_warning_l (<class 'custom_components.goodwe.sensor.InverterSensor'>) took 1.095 seconds. Please create a bug report at https://github.com/mletenay/home-assistant-goodwe-inverter/issues
	Line 669202: 2023-12-24 00:28:09.814 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.function_bit (<class 'custom_components.goodwe.sensor.InverterSensor'>) took 3.091 seconds. Please create a bug report at https://github.com/mletenay/home-assistant-goodwe-inverter/issues
```

hemiceni, 2023-12-25:
I just found some more odd behaviour with further testing.

Setting general mode usually works ok.
However when switching specifically from off-grid mode > general mode I experience the following issue:

-Neither battery nor PV supply house load
-House becomes powered 100% from grid
-No output at all from PV

If i switch modes back from general mode >i off-grid mode > general mode 
then general mode behaves as it should.

aldert-b, 2024-02-21:
Hi,
I have the same issue as JoseMadbird. I have a Goodwe GW6000EH and I want to charge my batteries starting from 12:00 at noon to prevent the possibility the voltage increases to values > 253V. For that I want to set the working mode to Eco mode when it is a sunny day and discharge at 1% in the morning. On cloudy days the mode should be General to be able to charge the battery as much as possible. But when changed from General to Eco mode, the selected time frames are deselected and Eco mode is behaving as general mode. I am using the official Goodwe version for home assistant and I was considering to change to the HACS version for this issue, but if I understand correct, the issue is not in the Goodwe firmware and switching to HACS version would not solve my issue.

JoseMadbird, 2024-02-21:
> Hi, I have the same issue as JoseMadbird. I have a Goodwe GW6000EH and I want to charge my batteries starting from 12:00 at noon to prevent the possibility the voltage increases to values > 253V. [...]

Hi, for me now is solved, i did 2 things, open a ticket in goodwe, explaining the issue, and the upgrade my inverter, in parallel, in my homeassistant, i´ve update Goodwee integration to all possible ones offered by platform (experimental included).
Now, i have more options in HA, and i can select in "inverter operation mode" - "Eco Charge Mode" and also select de "ECO mode power", when i do an automation, at the moment y select eco charge mode, automatically change and start to charge, with the % stablished in "eco mode power"
[image]

JoseMadbird, 2024-02-21:
[image]

aldert-b, 2024-03-06:
Hi JoseMadbird. Thanks for your reply. I did the same steps (contact Goodwe support and install experimental Goodwe version) now it works when I want to change the mode from General to Eco discharge mode. 
After I changed the mode to Eco discharge mode I saw in the SolarGo app that it generated 4 timeframes. Battery Power Group1-4. The first one is enabled with timeframe 24/7 and with 1% of discharge. Exactly what I want. I have tested to change the Eco Mode Power from 1 to 10, but I did not see it getting updated in the SolarGo app nor did I see the discharge increase.
Not sure what Eco Mode SoC setting is doing. Charge to this value when in Eco charge mode?