mletenay/home-assistant-goodwe-inverter #233: PV3 and PV4 dont show

gigi2485, 2023-12-23:
Inverter 15k et:
the entities for PV3 and PV4 are not displayed for me.
In SolarGo they are displayed as normal.
[image]
[image]

The current for the MPPT is also specified in watts 
[image]

thx

mletenay, 2023-12-31:
Indeed. 
There is a copy/paste mistake in the library using power instead of current. I'll fix it.
As for the pv3/4, there is problem as well (actually a confusion of pv and mppt values).
It would be great if you could provide the raw responses from the logs so I can add them to unit tests.

mletenay, 2024-02-06:
Fixed in v0.9.9.14

closed 2024-02-06 by mletenay