mletenay/home-assistant-goodwe-inverter #236: Power Limit question

roccolord, 2023-12-28:
Hello I have a GW15KT-DT and in SolorGo I can enable/disable power limit and set a percentage of the limit.

Setting limit to 4% and untoggle the power limit (so that it is disabled) in SolarGo displays the limit to be 4% in HA, but there is no toggle in HA.

How should I handle this in HA, setting the limit to 100% is that what SolarGo does? Probably not since the 4% is displayed in HA.

Any advice?

BR
Robert

arkovda, 2024-01-31:
Hi,
I have a GW10K-ET I Believe @roccolord has the same problem as I with the DOD. 
I can set it to 0-100 but it only works between 0-90, can't set it to 95. 
I know It's low but should work, and I have been asked to run it full cycle since it's been strange after adding a cell. 
/Arkovda

mletenay, 2024-02-14:
There is explicit safeguard to max 90% in the code, but I do not remember why I put it there :-)
I will remove it and allow full 0-100 range.

arkovda, 2024-02-15:
Thank you for telling me. I didn’t know where the problem was. For some reasons the problem persist even when I removed all of the code.

arkovda, 2024-02-21:
As far as I can tell, it seems to be working fine now
Thank you!

closed 2024-04-13 by mletenay