mletenay/home-assistant-goodwe-inverter #238: The special operating modes "Eco charge mode" and "Eco discharge mode" are not loading.

jirimartchus, 2023-12-29:
Hi, I have GoodWe GW10K-ET inverter firmware: 04029-07-S11 / 02041-22-S00. GoodWe Inverter (experimental) is integrated into HomeAssistant version: 2023.12.4.
When the integration is displayed, the special operating modes "Eco charge mode" and "Eco discharge mode" are missing. I have tried versions of GoodWe Inverter (experimental) from 0.9.9.9 to 0.9.9.13. It does not install the entities for these special modes. I'm stumped. Can you please advise?

2023-12-29 16:09:52.800 DEBUG (MainThread) [goodwe.protocol] Received: aa55f70306170c1d1009341dd9
2023-12-29 16:09:52.809 DEBUG (MainThread) [goodwe.protocol] Sending: f703b798000136c7
2023-12-29 16:09:53.635 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030200007051
2023-12-29 16:09:53.661 DEBUG (MainThread) [goodwe.protocol] Sending: f703b12c00017669
2023-12-29 16:09:53.707 WARNING (zeroconf-ServiceBrowser-_googlecast._tcp-119) [pychromecast.dial] Failed to determine cast type for host <unknown> (HTTP Error 403: Forbidden) (services:{ServiceInfo(type='mdns', data='Android-TV-2e3c977bb1edbd46374247ca854989e2._googlecast._tcp.local.')})
2023-12-29 16:09:53.708 DEBUG (MainThread) [goodwe.protocol] Received: aa55f70302000f3055
2023-12-29 16:09:53.739 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2023-12-29 16:09:53.961 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173b5500001400640000c697
2023-12-29 16:09:53.962 DEBUG (MainThread) [custom_components.goodwe.select] Could not read inverter operation mode
2023-12-29 16:09:53.962 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2023-12-29 16:09:54.211 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173b5500001400640000c697
2023-12-29 16:09:54.211 DEBUG (MainThread) [custom_components.goodwe.number] Could not read inverter setting eco_mode_power
2023-12-29 16:09:54.212 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2023-12-29 16:09:54.461 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173b5500001400640000c697
2023-12-29 16:09:54.462 DEBUG (MainThread) [custom_components.goodwe.number] Could not read inverter setting eco_mode_soc

lampra1, 2024-01-02:
Probably a firmware issue. Mine works fine.
Firmware: 04029-10-S11 / 02041-25-S00.
As for firmware upgrade

mletenay, 2024-01-08:
Firmware 22 seems to bring new value 0x55 as "undefined" eco mode, fix is in the works ...

jirimartchus, 2024-01-09:
> Firmware 22 seems to bring new value 0x55 as "undefined" eco mode, fix is in the works ...

jirimartchus, 2024-01-09:
> Firmware 22 seems to bring new value 0x55 as "undefined" eco mode, fix is in the works ...

Thank you for the answer...

closed 2024-01-09 by jirimartchus