mletenay/home-assistant-goodwe-inverter #241: It Cannot Access and Write to the Inverter GW6000ES20

Sascha883, 2024-01-04:
After firmware update to 060608 and ARM 08.362

The GW6000ES20 is only displayed as 04048 in Home Assistant and it cannot access and write to the inverter.

2024-01-04 00:46:05.415 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.224 seconds (success: True)
2024-01-04 00:46:18.494 DEBUG (MainThread) [goodwe] Probing inverter at 192.168.0.168.
2024-01-04 00:46:18.497 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241
2024-01-04 00:46:18.564 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001824c00060006080000000000000000000000000000000000000000000000000000353630303045534e3233335730303938000000000000000000000000000000000230323032302d30302d53000808cd
2024-01-04 00:46:18.566 DEBUG (MainThread) [goodwe] Detected ET/EH/BT/BH/GEH inverter , S/N:56000ESN233W0098.
2024-01-04 00:46:18.567 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2024-01-04 00:46:18.570 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1
2024-01-04 00:46:18.634 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703420079122a0000353630303045534e3233335730303938475736303030455332300006000619ce0008016a30343034382d30362d53303630323032302d30382d5330311e3d
2024-01-04 00:46:18.637 DEBUG (MainThread) [goodwe] Connecting to ET family inverter at 192.168.0.168.
2024-01-04 00:46:18.638 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2024-01-04 00:46:18.641 DEBUG (MainThread) [goodwe.protocol] Sending: f70388b800213ac1
2024-01-04 00:46:18.704 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703420079122a0000353630303045534e3233335730303938475736303030455332300006000619ce0008016a30343034382d30362d53303630323032302d30382d5330311e3d
2024-01-04 00:46:18.705 DEBUG (MainThread) [goodwe] Connected to inverter GW6000ES20, S/N:56000ESN233W0098.
2024-01-04 00:46:18.708 DEBUG (MainThread) [goodwe.protocol] Sending: f703891c007d7ae7
2024-01-04 00:46:18.794 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703fa180104002e1100000000000000000000000000000000000000000000000000000000000000000000000008d800031388000000100000000000000000000000000000000000000000000100000010fffff3b5000000600000006208d1000313880001000000000000000000000000000000000000000000000000000000000000051000000304000003f80000000000000c6100000101000000d300000e10ffff021d00070000002500020000000200010201000000000000028d00000000000016e200000717000000001af100010000fc69001200001a480000000016c2000000000000000000000000000000000000000001c8000300c80000ae10
2024-01-04 00:46:18.798 DEBUG (MainThread) [goodwe.protocol] Sending: f70390880018fc7c
2024-01-04 00:46:18.864 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7033000ff00000000006500000001100000300064000100010120000000070000000000000000000000000000000000000000d1d5
2024-01-04 00:46:18.869 DEBUG (MainThread) [goodwe.protocol] Sending: f7038ca0002dbbf3
2024-01-04 00:46:18.933 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7035a00030000000a01110001fa01fcfdfc02f2ff06bdbd58c71caa34b9d413880000000000000000fffffa01fffffcfdfffffc02fffff2ff000002f0000001f5000001d7000006bdfffff8e1fffffc3cfffffb6dfffff08a000200055db4
2024-01-04 00:46:18.935 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.229 seconds (success: True)
2024-01-04 00:46:18.948 DEBUG (MainThread) [custom_components.goodwe.switch] Could not read load control switch value
2024-01-04 00:46:18.949 DEBUG (MainThread) [goodwe.protocol] Sending: f703b090000337b0
2024-01-04 00:46:19.267 DEBUG (MainThread) [goodwe.protocol] Received: aa55f70306180104002e12ac54
2024-01-04 00:46:19.328 DEBUG (MainThread) [goodwe.protocol] Sending: f703b996000155ec
2024-01-04 00:46:19.388 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030227106a6d
2024-01-04 00:46:19.395 DEBUG (MainThread) [goodwe.protocol] Sending: f703b798000136c7
2024-01-04 00:46:19.454 DEBUG (MainThread) [goodwe.protocol] Received: aa55f703020005b052
2024-01-04 00:46:19.463 DEBUG (MainThread) [goodwe.protocol] Sending: f703b12c00017669
2024-01-04 00:46:19.527 DEBUG (MainThread) [goodwe.protocol] Received: aa55f70302001ef059
2024-01-04 00:46:19.539 DEBUG (MainThread) [goodwe.protocol] Sending: f703b99b0004042c
2024-01-04 00:46:19.603 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030800000500fc185500070e
2024-01-04 00:46:19.605 DEBUG (MainThread) [custom_components.goodwe.select] Could not read inverter operation mode
2024-01-04 00:46:19.608 DEBUG (MainThread) [goodwe.protocol] Sending: f703b99b0004042c
2024-01-04 00:46:19.666 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030800000500fc185500070e
2024-01-04 00:46:19.682 DEBUG (MainThread) [custom_components.goodwe.number] Could not read inverter setting eco_mode_power
2024-01-04 00:46:19.686 DEBUG (MainThread) [goodwe.protocol] Sending: f703b99b0004042c
2024-01-04 00:46:19.743 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030800000500fc185500070e
2024-01-04 00:46:19.745 DEBUG (MainThread) [custom_components.goodwe.number] Could not read inverter setting eco_mode_soc
2024-01-04 00:46:23.944 DEBUG (MainThread) [goodwe.protocol] Sending: f703891c007d7ae7

mletenay, 2024-01-04:
What is not working ? 
The logs are mostly fine.

Sascha883, 2024-01-05:
Hi,

In the first picture is the inverter from Goodwe GW6000ES20

The menus are missing during Configuration
Inverter Operation Mode, Eco Mode Power and Eco Mode SoC

[image]

In the second picture is the inverter from Goodwe GW8K-ET. All entities are found here

Just not with the first inverter. All Inverters are Connected to the same Network via LAN

[image]

[image]

mletenay, 2024-01-08:
I see now.
There is probably some problem decoding Eco Mode values in ES G2 inverters, I need to figure out what.

Sascha883, 2024-01-09:
Hi, thanks for the information

I created a log file with debug mode and found the entry "Could not read load control switch value" there.

[home-assistant_2024-01-09T21-09-24.327Z.log](https://github.com/mletenay/home-assistant-goodwe-inverter/files/13879136/home-assistant_2024-01-09T21-09-24.327Z.log)

estimadarocha, 2024-01-10:
[image]

the options are in grayscale!

madeupusername123, 2024-01-27:
Hello, had to do a firmware update to get a new lan dongle working; running Goodwe HA + HACS Experimental for commanding inverter to charge/discharge etc. Firmware update did similar function as above. I am running GW5000ES20 (ES G2). Having the same issue as above.

estimadarocha, 2024-02-01:
@mletenay is there anything we can do to help you debugging this?

mletenay, 2024-02-02:
It should be already fixed in the library, I just need to make a release and update HA integration ...

nbalcaen, 2024-02-04:
@mletenay any idea when the release will be available?

I'm having the same issue with Goodwe GW3600SBP20.

nbalcaen, 2024-02-06:
@mletenay thanks for the update! Now I'm struggling with the next problem, the Eco charge and Eco discharge doesn't work. Any idea how to fix it. 

[image]

mletenay, 2024-02-06:
Fixed in v0.9.9.14

shubsd, 2024-03-28:
Hi @mletenay 
For a new install, I'm getting the same issue with the greyed out box.  Any advice would be really appreciated

[image]
[image]