mletenay/home-assistant-goodwe-inverter #245: PV1 Voltage = PV2 Voltage and PV2-Currunt alway 0A @ 3-Stringed GW25ET

hbpv, 2024-01-25:
Hi

We've two major Problems with values:

1. PV1 VOltage is alway the same as PV2 Voltage, but the Strings are strongly different. Please see the screenshot. Is it possible that there is an copy-paste mistake in polling the same values inside the script?

2. PV2 Current stay always at 0. As this String is producing >3kwh, this is wrong.

Thanks and best,
Max

[image]
[image]

closed 2024-01-25 by mletenay

hbpv, 2024-01-25:
Okay, so the entities we get are wrong?
The Inverter has 6 PV Ports
1+2 = PV1
3+4 = PV2
5+6 = PV3

Our first String is on Input 1 (PV1-1)
Our second String is on Input 3 (PV2-1)
Our third String is on Input 5 (PV3-1)

In the home assistant integration, the way your argument, we should get the entities PV1 + PV3 + PV5.
But it only shows 1,2,3. Also, when they have the same Voltage, the other values should be the same regarding your idea, but they're not.

Please see attached the Screenshot for the integration.

[image]
[image]

hbpv, 2024-01-25:
Here you’ve got the sems values, where it shows correctly 3 different strings and their voltages etc

[image]

mletenay, 2024-01-25:
Well, what I know is there is no copy/paste mistake in the code - see for yourself: https://github.com/marcelblijleven/goodwe/blob/master/goodwe/et.py
The rest is responsibility of inverter what values it provides.

mletenay, 2024-01-25:
... and the current integration does not have the latest library code yet, so you may not see the pv5+ sensors.
If you are able, try the library directly and run inverter_check.py to see all values inverter reports.

hbpv, 2024-01-25:
Okay.. And now? How could I further help to get this resolved?

As you can see in the SEMS Screenshot, there are 3 different values. So maybe we should check, why the integration/inventer is not resolving the ports that are connected (PV1-1 = PV1) + (PV2-1 = PV3) + (PV3-1 = PV5). Maybe wrong pull of modbus registers? In the et.py i see a lot more possible PV's up to 16.

Can i somewhere check/log which values are provided? Maybe we can there find the problem.

hbpv, 2024-01-25:
> ... and the current integration does not have the latest library code yet, so you may not see the pv5+ sensors. If you are able, **try the library directly** and run inverter_check.py to see all values inverter reports.

Okay.. and how to update our library? Could give me a short how to? Thanks

mletenay, 2024-01-25:
I will release next version of the library probably over the weekend and then adjust the HA integration as well.

hbpv, 2024-01-25:
> I will release next version of the library probably over the weekend and then adjust the HA integration as well.

Okay.. i just installed the moist recent version of hacs-goodwe from you, but problem persists. Is this the same release channel?

hbpv, 2024-01-25:
For your release, please be aware that with hacs integration:

MPTT entities are working fine (all 3 with correct values) BUT MPPTX Current are W(Watt) instead of A(ampere), see screenshot
[image]

And:

As i showed you before, all (exept PV4 is missing completly??) PV's are showing, but with wrong Voltages:

Starting with

(MPPT1) PV1 - Voltage and Current OK
(MPPT1) PV1 - Voltage and Current OK
(MPPT1) PV1 - Voltage and Current OK

[image]