mletenay/home-assistant-goodwe-inverter #252: No inverter mode/Eco soc with latest versions 5048D-ES

robotz0mbie, 2024-02-11:
Hi, firstly thanks for making a great plugin, I have been using it for a long time to control my goodwe 5048D_ES system.
I was running v 9.9.9.7 with no problems prior, stuck with an old version due to issues goodwe had updating the inverter firmware, which turned out to be related to a weak wifi signal somehow stopping the process. I sorted that out recently and finally have the latest ARM (v20) and DSP (2626?) on the inverter. So great, I could finally upgrade to the newest plugin and get the ECO mode SOC features etc added after 9.9.9.7.

I noticed after updating to the newest HA (2024.2?), the inverter control function became unavailable. (still running plugin code 9.9.9.7), in the log I get the following:

`2024-02-12 10:26:02.672 DEBUG (MainThread) [goodwe] Connecting to ES family inverter at 192.168.1.55.
2024-02-12 10:26:02.678 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2024-02-12 10:26:02.698 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241
2024-02-12 10:26:03.082 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001824d3233323345475735303438442d45532331302020202020202020202020202039353034384553553232335730303131333630303431302d30343032352d3236493431302d30323033342d323001120f
2024-02-12 10:26:03.086 DEBUG (MainThread) [goodwe] Connected to inverter GW5048D-ES, S/N:95048ESU223W0011.
2024-02-12 10:26:03.091 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2024-02-12 10:26:03.841 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0c28003a020c34002f0202180002005000e601c60064006400001f001f5d030000010916001e0003138a0100680000029f138a01020192000000000001c092000031f7004300430001764b02ad01003000000000010000001d01100800000018020c0a1a020000000000000000000000000000000000000000000000000000ebcc0000ea170200000000000012ac
2024-02-12 10:26:03.845 DEBUG (MainThread) [custom_components.goodwe] Finished fetching GoodWe data in 0.757 seconds (success: True)
2024-02-12 10:26:03.860 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-02-12 10:26:03.868 ERROR (MainThread) [homeassistant.components.select] Error while setting up goodwe platform for select
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 344, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/goodwe/select.py", line 53, in async_setup_entry
    OPERATION_MODE.options = [
    ^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 4, in __setattr__
dataclasses.FrozenInstanceError: cannot assign to field 'options'
2024-02-12 10:26:04.299 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000000000000001000100000001000a013b022900f0008001e0000a000f0000000000000064024003e8001e00ea0ed8000000000000023f000000070000000000000001038403e801e0000a000000220c43
2024-02-12 10:26:04.311 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-02-12 10:26:04.649 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000000000000001000100000001000a013b022900f0008001e0000a000f0000000000000064024003e8001e00ea0ed8000000000000023f000000070000000000000001038403e801e0000a000000220c43`

Ok, I thought maybe there is an issue with the old plugin and the new HA. So I updated to the newest plugin (9.9.9.14), however I still get the same issue, no inverter control, and no Eco mode SOC settings etc. In the log I get the following: 

`2024-02-12 10:32:03.914 DEBUG (MainThread) [goodwe] Connecting to ES family inverter at 192.168.1.55.
2024-02-12 10:32:03.915 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2024-02-12 10:32:03.919 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241
2024-02-12 10:32:04.528 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001824d3233323345475735303438442d45532331302020202020202020202020202039353034384553553232335730303131333630303431302d30343032352d3236493431302d30323033342d323001120f
2024-02-12 10:32:04.539 DEBUG (MainThread) [goodwe] Connected to inverter GW5048D-ES, S/N:95048ESU223W0011.
2024-02-12 10:32:04.543 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2024-02-12 10:32:04.660 WARNING (MainThread) [homeassistant.bootstrap] Waiting on integrations to complete setup: goodwe
2024-02-12 10:32:04.994 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0c22003c020c1a00310202180002005000f001d00064006400002100215d030000010909001b0004138b01006700000260138b0102019c000000000001c093000031f7004400440001764c027501003000000000010000002f01100800000018020c0a20030000000000000000000000000000000000000000000000000000ebce0000ea1702000000000000124c
2024-02-12 10:32:04.999 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.460 seconds (success: True)
2024-02-12 10:32:05.243 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-02-12 10:32:05.903 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000000000000001000100000001000a013b022900f0008001e0000a000f0000000000000064024003e8001e00ea0ed8000000000000023f000000070000000000000001038403e801e0000a000000220c43
2024-02-12 10:32:05.909 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-02-12 10:32:06.358 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000000000000001000100000001000a013b022900f0008001e0000a000f0000000000000064024003e8001e00ea0ed8000000000000023f000000070000000000000001038403e801e0000a000000220c43
2024-02-12 10:32:06.366 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-02-12 10:32:06.929 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000000000000001000100000001000a013b022900f0008001e0000a000f0000000000000064024003e8001e00ea0ed8000000000000023f000000070000000000000001038403e801e0000a000000220c43
2024-02-12 10:32:06.936 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-02-12 10:32:07.334 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000000000000001000100000001000a013b022900f0008001e0000a000f0000000000000064024003e8001e00ea0ed8000000000000023f000000070000000000000001038403e801e0000a000000220c43
2024-02-12 10:32:07.341 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2024-02-12 10:32:07.403 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173b007f0000383453454d22
2024-02-12 10:32:07.405 DEBUG (MainThread) [custom_components.goodwe.select] Could not read inverter operation mode
Traceback (most recent call last):
  File "/config/custom_components/goodwe/select.py", line 60, in async_setup_entry
    eco_mode = await inverter.read_setting("eco_mode_1")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/goodwe/es.py", line 226, in read_setting
    return setting.read_value(response)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/goodwe/sensor.py", line 528, in read_value
    raise ValueError(f"{self.id_}: SoC value {self.soc} out of range.")
ValueError: eco_mode_1: SoC value 14388 out of range.
2024-02-12 10:32:07.417 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2024-02-12 10:32:07.483 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173b007f0000383453454d22
2024-02-12 10:32:07.487 DEBUG (MainThread) [custom_components.goodwe.number] Could not read inverter setting eco_mode_power
2024-02-12 10:32:07.491 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2024-02-12 10:32:07.562 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173b007f0000383453454d22
2024-02-12 10:32:07.565 DEBUG (MainThread) [custom_components.goodwe.number] Could not read inverter setting eco_mode_soc`

I also noticed the diagnostics for the new version seem to be reported the incorrect ARM version (v14) rather than V20 which I have confirmed with goodwe and another utility.

Any idea what might be going on here or how I could fix it ?

cheers

mletenay, 2024-02-14:
v14 is raw hex value - i.e. 20 decimal, so it's ok.
The trouble is that 14388 SoC reported by the inverter - there is a check for 0-100, so something is odd there, need to inverstigate.

mletenay, 2024-02-14:
The 14388 of SoC makes no sense, it smell like if it some odd non-initialized value ?!
You should try to set the EcoMode with the SolarGo application first (to ensure the register are properly initialized ??).
And then restart the HA and check again

robotz0mbie, 2024-02-15:
thanks for the reply ! 
I will try that and get back to you. I wonder, is there a way I can access the older version in HACS for rolling back? at the moment the oldest HACS shows is 9.9.9.10, so if I change I need to do a full HA restore to get the 9.9.9.7 back which is working well!!

tinuva, 2024-02-15:
If it helps, I noticed that my "inverter mode" is also greyed out on the latest version. However my ECO SoC control is there.

I noticed that this issue comes and goes and I haven't really been able to track down what triggers this. It started around the time I logged #219

mletenay, 2024-02-15:
@robotz0mbie Yes, the v0.9.9.14 requires HA 2024.2, I've just forgot to increase required version in manifest.
As for the rollback to old version, well never tried/needed that so I'm not sure.

@tinuva There is kind of explanation of the instabilities.
During startup, the integration tries to figure out supported features by invoking couple of commands.
If the command fails - it is assumed it is not supported by the inverter and gets greyed out or hidden.
The trouble is with the unreliable nature of the UDP protocol.
There are 3 retries in the communication code, but if you encounter some network instability during startup, it may lead to false negatives.
That's why after restart it sometimes appears and sometime disappears, depending if the responses cam in time/correct.

robotz0mbie, 2024-02-15:
hmmm so I did some more testing. I installed a fresh HA instance on another Rpi, installed HACS and the latest version of the plugin. Disabled the plugin on my main HA, out of the box I get the same error on the new install (no operation mode, eco soc etc, 61/65 entities)

2024-02-16 09:16:54.503 DEBUG (MainThread) [goodwe] Connecting to ES family inverter at 192.168.1.55.
2024-02-16 09:16:54.504 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2024-02-16 09:16:54.506 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241
2024-02-16 09:16:55.178 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001824d3233323345475735303438442d45532331302020202020202020202020202039353034384553553232335730303131333630303431302d30343032352d3236493431302d30323033342d323001120f
2024-02-16 09:16:55.180 DEBUG (MainThread) [goodwe] Connected to inverter GW5048D-ES, S/N:95048ESU223W0011.
2024-02-16 09:16:55.183 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2024-02-16 09:16:55.748 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0d00000b020cba000a0202000002005000d2001d0065006400000e000e5d03000001091e00160003138b010067000001de138b01020118000000000001c46b00003256003900260001788201ea0100300000000001000000250100080000001802100910340000000000000000000000000000000000000000000000000000edb20000ec1802000000000000111e
2024-02-16 09:16:55.751 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.570 seconds (success: True)
2024-02-16 09:16:55.762 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-02-16 09:16:56.228 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000000000000001000100000001000a013b022900f0008001e0000a000f0000000000000064024003e8001e00ea0ed8000000000000023f000000070000000000000001038403e801e0000a000000220c43
2024-02-16 09:16:56.232 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-02-16 09:16:56.557 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000000000000001000100000001000a013b022900f0008001e0000a000f0000000000000064024003e8001e00ea0ed8000000000000023f000000070000000000000001038403e801e0000a000000220c43
2024-02-16 09:16:56.561 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-02-16 09:16:56.963 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000000000000001000100000001000a013b022900f0008001e0000a000f0000000000000064024003e8001e00ea0ed8000000000000023f000000070000000000000001038403e801e0000a000000220c43
2024-02-16 09:16:56.967 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-02-16 09:16:57.302 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000000000000001000100000001000a013b022900f0008001e0000a000f0000000000000064024003e8001e00ea0ed8000000000000023f000000070000000000000001038403e801e0000a000000220c43
2024-02-16 09:16:57.307 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2024-02-16 09:16:57.367 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173b007f0000383453454d22
2024-02-16 09:16:57.368 DEBUG (MainThread) [custom_components.goodwe.select] Could not read inverter operation mode
Traceback (most recent call last):
  File "/config/custom_components/goodwe/select.py", line 60, in async_setup_entry
    eco_mode = await inverter.read_setting("eco_mode_1")
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/goodwe/es.py", line 226, in read_setting
    return setting.read_value(response)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/goodwe/sensor.py", line 528, in read_value
    raise ValueError(f"{self.id_}: SoC value {self.soc} out of range.")
ValueError: eco_mode_1: SoC value 14388 out of range.
2024-02-16 09:16:57.381 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2024-02-16 09:16:57.442 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173b007f0000383453454d22
2024-02-16 09:16:57.444 DEBUG (MainThread) [custom_components.goodwe.number] Could not read inverter setting eco_mode_power
2024-02-16 09:16:57.447 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2024-02-16 09:16:57.508 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173b007f0000383453454d22
2024-02-16 09:16:57.509 DEBUG (MainThread) [custom_components.goodwe.number] Could not read inverter setting eco_mode_soc
2024-02-16 09:17:00.969 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2024-02-16 09:17:01.468 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0cc4000c020cce000a0202000002005000d200190065006400000e000e5d03000001091b00170003138a010067000001da138a01020117000000000001c46b00003256003900260001788201ee01003000000000010000002f01000800000018021009103a0000000000000000000000000000000000000000000000000000edb20000ec180200000000000011fd
2024-02-16 09:17:01.471 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.503 seconds (success: True)
2024-02-16 09:17:05.970 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245

I then tried @mletenay idea of initialising with the inverter in eco mode. I disabled the plugin on the new install, enabled on the old install. Set the inverter to eco discharge mode. Disabled the plugin on the old, and re-initialised the plugin on the new install. Showed up in discharge mode as expe [... cut ...]
2024-02-16 09:19:28.965 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2024-02-16 09:19:29.553 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0cc0000c020ce2000a0201fd0002005000d2006e0065006400000e000e5d020000010922003202c71389010067000001d0138901020117000000000001c46b000032560039002600017882049601003001000000010000037c01080850000018021009131a0000000000000000000000000000000000000000000000000000edb20000ec18020000000000001470
2024-02-16 09:19:29.556 DEBUG (MainThread) [custom_componen

mysterious !

robotz0mbie, 2024-02-16:
to properly test the suggestion, I set the inverter to eco mode / discharge via SolarGo, then re-initialised the new version. 
This time, there are no errors in the log, and the inverter mode control is available. However, no Eco mode SoC settings available. I also didnt see Eco mode SoC settings in solarGo... maybe they just arent available with this inverter? 

2024-02-16 17:42:50.791 DEBUG (MainThread) [goodwe] Connecting to ES family inverter at 192.168.1.55.
2024-02-16 17:42:50.791 DEBUG (MainThread) [goodwe.inverter] Creating lock instance for current event loop.
2024-02-16 17:42:50.794 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241
2024-02-16 17:42:51.062 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001824d3233323345475735303438442d45532331302020202020202020202020202039353034384553553232335730303131333630303431302d30343032352d3236493431302d30323033342d323001120f
2024-02-16 17:42:51.064 DEBUG (MainThread) [goodwe] Connected to inverter GW5048D-ES, S/N:95048ESU223W0011.
2024-02-16 17:42:51.067 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2024-02-16 17:42:51.630 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0e450028020e45002202021300020050010e01950065006400006300635d02000001095b00ba0e98138901006c000003261389010201a0000000000001c4990000325e0067004f000178ab11bc0100300100000001000003e4010808500000180210112a340000000000000000000000000000000000000000000000000000ee3f0000ec1a020000000000001454
2024-02-16 17:42:51.633 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.569 seconds (success: True)
2024-02-16 17:42:51.645 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-02-16 17:42:51.980 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000173b00320001000100000001000a013b022900f0008001e0000a000f0000000000000064024003e8001e00ea0ed8000000000000023f000000070003000000000001038403e801e0000a000000220cca
2024-02-16 17:42:51.986 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-02-16 17:42:52.439 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000173b00320001000100000001000a013b022900f0008001e0000a000f0000000000000064024003e8001e00ea0ed8000000000000023f000000070003000000000001038403e801e0000a000000220cca
2024-02-16 17:42:52.444 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-02-16 17:42:52.844 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000173b00320001000100000001000a013b022900f0008001e0000a000f0000000000000064024003e8001e00ea0ed8000000000000023f000000070003000000000001038403e801e0000a000000220cca
2024-02-16 17:42:52.849 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-02-16 17:42:53.304 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000173b00320001000100000001000a013b022900f0008001e0000a000f0000000000000064024003e8001e00ea0ed8000000000000023f000000070003000000000001038403e801e0000a000000220cca
2024-02-16 17:42:53.308 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2024-02-16 17:42:53.369 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173bff7f0032000000008a4f
2024-02-16 17:42:53.373 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2024-02-16 17:42:53.434 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173bff7f0032000000008a4f
2024-02-16 17:42:53.436 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2024-02-16 17:42:53.497 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173bff7f0032000000008a4f
2024-02-16 17:42:53.499 DEBUG (MainThread) [goodwe.protocol] Sending: f703b9bb00068427
2024-02-16 17:42:53.560 DEBUG (MainThread) [goodwe.protocol] Received: aa55f7030c0000173bff7f0032000000008a4f
2024-02-16 17:42:56.615 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2024-02-16 17:42:57.025 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0e310029020e33002302021200020050010e01840065006400006300635d02000001096000ba0e9f138901006c0000031e1389010201a1000000000001c4990000325e0067004f000178ab11bd0100300100000001000003e4050808500000180210112a380000000000000000000000000000000000000000000000000000ee3f0000ec1a02000000000000142c
2024-02-16 17:42:57.028 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.413 seconds (success: True)
2024-02-16 17:43:01.616 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2024-02-16 17:43:02.004 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0e20002b020e37002302021200020050010e01720065006400006300635d02000001096000b70e7c138901006c000003011389010201a2000000000001c4990000325e0067004f000178ab117d0100300100000001000003e4050808500000180210112b020000000000000000000000000000000000000000000000000000ee3f0000ec1a020000000000001358
2024-02-16 17:43:02.007 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.392 seconds (success: True)
2024-02-16 17:43:06.615 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2024-02-16 17:43:07.045 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0e1a002c020e12002402021200020050010e015e0065006400006300635d02000001095e00b70e5e138901006c0000031e1389010201a5000000000001c4990000325e0067004f000178ab117c0100300100000001000003e4050808500000180210112b070000000000000000000000000000000000000000000000000000ee3f0000ec1a02000000000000131f
[... cut ...]
2024-02-16 17:43:17.062 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.448 seconds (success: True)
2024-02-16 17:43:21.615 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2024-02-16 17:43:22.129 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001868c0df5002f020df1002502021200020050010e01480065006400006300635d02000001095900b70e91138901006b000002fa1389010201a8000000000001c49a0000325e0068004f000178ab11870100300100000001000003e5010808500000180210112b160000000000000000000000000000000000000000000000000000ee3f0000ec1a0200000000000015e9

closed 2024-02-16 by robotz0mbie