mletenay/home-assistant-goodwe-inverter #258: Error when trying to add Goodwe

tranceblast, 2024-03-13:
Its not possible to connect to Goodwe 15kw with WiFi 2.0
[image]

mletenay, 2024-03-15:
Does the SolarGo/PvMaster mobile app work ?

linusst, 2024-04-02:
Hi,
I have the same problem. Yes SolarGo and SEMS is working.

This is the log i get:

Logger: aiohttp.server
Source: /usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py:421
First occurred: April 1, 2024 at 22:27:58 (7 occurrences)
Last logged: 13:42:52

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_protocol.py", line 452, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_app.py", line 543, in _handle
    resp = await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/aiohttp/web_middlewares.py", line 114, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 91, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 25, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 79, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 235, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 31, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 68, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 72, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 197, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 72, in wrapper
    result = await method(view, request, data, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 111, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 359, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 405, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 502, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/goodwe/config_flow.py", line 97, in async_step_user
    inverter = await connect(host=host, family=model_family, retries=10)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/goodwe/__init__.py", line 51, in connect
    return await discover(host, timeout, retries)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/goodwe/__init__.py", line 71, in discover
    response = await DISCOVERY_COMMAND.execute(host, timeout, retries)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 154, in execute
    await response_future
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 1228, in _read_ready
    data, addr = self._sock.recvfrom(self.max_size)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ConnectionRefusedError: [Errno 111] Connection refused

micros01, 2024-04-09:
Also have the same problem, newly installed 15kw hybrid inverter.
Looks like the inverter doesn't respond to UDP 8899 at all(tested with nmap), updated to latest DSP and ARM firmware(according to SolarGo)
DSP: 07.7068
ARM: 08.401

mletenay, 2024-04-10:
How do you then connect SolarGo ? 
Via bluetooth?
If not, can you sniff network traffic ?

micros01, 2024-04-10:
I'm using Bluetooth to connect SolarGo.
I've never been able to see the Wifi the inverter is supposed to broadcast..havent found any setting where you can turn on/off this..maybe it's automatically disabled when the LAN module is installed?

mletenay, 2024-04-15:
Hmm, this seems to be the same issue as https://github.com/home-assistant/core/issues/114173
It looks like the new WIFI/LAN 2.0 adapters switched from UDP to (more standard) modbus/TCP.
And that one is not supported at the moment.

As mentioned in that other ticket, it should be doable, but will be a pain without real device to test with ...

micros01, 2024-04-23:
FYI..
I experimented with the modbus connection to my ET15, as far as I can tell it's read only but at least now I can get the statistics/values I need into Homeassistant, PV generation, battery and so on. 
Also seem to work fine adding the daily production to the Energy dashboard. 

I've added everything to my Github, maybe it could help someone else.(not used to Github but I think it should work :) )
https://github.com/micros01/GoodweHAmodbus

micros01, 2024-04-23:
OK, found this about the modbus registers: https://forum.iobroker.net/assets/uploads/files/1666333836822-goodwe_modbus_protocol_hybrid_et_eh_bh_bt__arm205-hv__v1.7-_-read-only_20200226-1.pdf and it states "Ready only"....which I interpreted to read only for the modbus but guess I was wrong. 

Then it should be possible to control charging and discharging the battery based on electrical price from the grid(which I get from another sensor in HA)? 
If you can hint on how to do that it would be awesome...

linusst, 2024-04-23:
Use this instead https://forum.iobroker.net/assets/uploads/files/1703579717824-arm.745.esg2.et30.modbus.protocol.map.20221231..v1.pdf?fbclid=IwZXh0bgNhZW0BMQABHXcRLb0ZpNyMluMOU1YbESC-qq7E6tqPzmVHqIo23sWL3AheL7OYA5a_fg_aem_AeJDg4Y2IXxYAnOrXnNez3tsuNQNVzB0cyxXopC87s3VG7X8rW4mRa4WvMJevb3suLo

mletenay, 2024-05-05:
Proper support for Modbus/TCP was added in v0.9.9.18.
When adding the integration, one can select either UDP (8899) or TCP (502).
Enjoy.

closed 2024-05-05 by mletenay