mletenay/home-assistant-goodwe-inverter #259: Grid Import Power Limit?

hansherlighed, 2024-03-18:
Hi,

First of, thx for an incredible integration!

When selecting "Peak Shaving mode" in the SEMS portal there's a setting called "Grid Import Power Limit" that controls when the batteries kick in and start providing power to help shave the consumption peak, It's value is in Watt. I cannot find this setting in Home assistant? If I choose "Peak shaving mode" in HA, what setting can i/should I use to specify the "Grid Import Power Limit"?

Thx.

hansherlighed, 2024-03-22:
did some more digging, and from I can see [here](https://github.com/OpenEMS/openems/blob/529095411164e9c9ca4754f25b5ec6be3e251425/io.openems.edge.goodwe/src/io/openems/edge/goodwe/common/AbstractGoodWe.java#L1519) it seems that the register ID is 47542 for this setting. I'm not knowledgeable in Python to try and implement this myself, but would be greatly appreciated if someone could implement this:-)

mletenay, 2024-04-13:
I've just added peak_shaving_power_limit and peak_shaving_soc settings values.
Once it is released you should be able to get/set those parameters with services

closed 2024-05-25 by mletenay

ABEC7, 2025-01-16:
> I've just added peak_shaving_power_limit and peak_shaving_soc settings values. Once it is released you should be able to get/set those parameters with services

I cant fins these values in the latest release. Are they coming in a future release or am i missing something?