mletenay/home-assistant-goodwe-inverter #261: GoodWe Inverter operation mode is not read from the inverter

jebeke65, 2024-03-20:
Hi,

Apparently the "GoodWe Inverter operation mode" does not fetch the mode from the invertor. Is this by design ?
It happens that I set the operation mode but that it is not 'accepted' by the inverter, which then remains in the previous state.
This gives a mismatch between the mode in HA and the mode in the inverter.

Jan

mletenay, 2024-05-03:
https://github.com/home-assistant/core/issues/116706

mletenay, 2024-05-05:
The "settings" values are not automatically refreshed by design, but since v0.9.9.18 all entities can be explicitly refreshed by _homeassistant.update_entity_ service.
If it is needed regularly, one can create and automation like this:
```
trigger:
  - platform: time_pattern
    minutes: "5"
condition: []
action:
  - service: homeassistant.update_entity
    target:
      entity_id: number.depth_of_discharge_on_grid
    data: {}
mode: single
```

closed 2024-05-05 by mletenay