mletenay/home-assistant-goodwe-inverter #267: [Feature Request] Add grid switch option

DrBlokmeister, 2024-04-06:
With negative energy prices arising again, I would like the option to switch my inverter on and off. The grid export option does nothing for me, even though the switch is set correctly in the SolarGo app. The option that does work is the Grid Switch control, where you can simply start and stop the inverter. Could you please implement this as a switch or maybe point to documentation on how to use the goodwe.set_parameter service and get the right parameter?

Thanks in advance!

mletenay, 2024-04-13:
I am not sure which switch do you mean.
You mean export limit switch ?

DrBlokmeister, 2024-04-14:
No. There is indeed also the switch that allows you to set an export limit, to limit the inverter to say 1000W, which is adjustable using an input number. This would be nice to have too, but this is not the switch I'm talking about. _(About this export limit, is it normal to set the limit to 1W, and still have the inverter deliver a minimum of 200W to the grid? My limiter only works down to ~200W. If I put the export limit to anywhere below that, it will just clamp to 200W. If I put the limit higher than 200W, for example to 400W, the delivery to the grid is ~400W, so that works. Is that normal operation?)_

In the SolarGo app there is the functionality to remotely shut down the inverter completely. It's basically an on/off switch for the inverter. I tried intercepting the package that is being sent to the inverter using a few wireshark-like apps on Android, but didn't really get anywhere.

TomEkk33, 2024-05-06:
@DrBlokmeister is it "DRED/Remote should down/RCR" switch in SolarGo? 
I have added it in switch.py diff against v 0.9.9.12. Maybe someone willing to add this to main branch...

```
Index: switch.py
===================================================================
--- switch.py	(revision 2)
+++ switch.py	(revision 3)
@@ -27,6 +27,30 @@
 )
 
 
+DRED_OFF = SwitchEntityDescription(
+    key="dred",
+    name="DRED/Remote Shutdown",
+    icon="mdi:transmission-tower-off",
+    entity_category=EntityCategory.CONFIG,
+    device_class=SwitchDeviceClass.SWITCH,
+)
+
 async def async_setup_entry(
     hass: HomeAssistant,
     config_entry: ConfigEntry,
@@ -52,6 +76,55 @@
         async_add_entities([entity])
 
 
+    # read current DRED from the inverter
+    try:
+        current_state = await inverter.read_setting("dred")
+        _LOGGER.debug("DRED read")
+    except (InverterError, ValueError):
+        # Inverter model does not support this feature
+        _LOGGER.debug("Could not read DRED value")
+    else:
+        entity = DredOFFSwitch(
+            device_info,
+            DRED_OFF,
+            inverter,
+            current_state == 1,
+        )
+        async_add_entities([entity])
+
+    
 class LoadControlSwitch(SwitchEntity):
     """Switch representation of inverter's 'Load Control' relay."""
 
@@ -87,3 +160,111 @@
         """Process update from entity."""
         status = await self._inverter.read_setting("load_control_switch")
         return status == 1
+
+class DredOFFSwitch(SwitchEntity):
+    """Switch representation of inverter's 'Fast Charging' relay."""
+
+    _attr_should_poll = False
+
+    def __init__(
+        self,
+        device_info: DeviceInfo,
+        description: SwitchEntityDescription,
+        inverter: Inverter,
+        current_is_on: bool,
+    ) -> None:
+        """Initialize the inverter operation mode setting entity."""
+        self.entity_description = description
+        self._attr_unique_id = f"{description.key}-{inverter.serial_number}"
+        self._attr_device_info = device_info
+        self._attr_is_on = current_is_on
+        self._inverter: Inverter = inverter
+
+    async def async_turn_on(self, **kwargs: Any) -> None:
+        """Turn the DRED ON - shutdown inverter"""
+        await self._inverter.write_setting("dred", 1)
+        self._attr_is_on = True
+        self.async_write_ha_state()
+
+    async def async_turn_off(self, **kwargs: Any) -> None:
+        """Turn the DRED OFF - Inverter ON."""
+        await self._inverter.write_setting("dred", 0)
+        self._attr_is_on = False
+        self.async_write_ha_state()
+
+    async def async_update(self) -> None:
+        """Process update from entity."""
+        status = await self._inverter.read_setting("dred")
+        return status == 1

DrBlokmeister, 2024-05-09:
No it's not this one. DRED is an option that you can check or uncheck. This is really a shutdown and start button that stops the production altogether. You can hear the internal relays click inside the inverter when you press the button. When you start it back up afterwards, there is some sort of cooldown before it really starts up again.

mletenay, 2024-05-09:
Which inverter model ?
Can you paste SolarGo screenshot ?

kokkie-d, 2024-05-12:
From the SolarGo app with my inverter and the effect, when you flip the switch. I would also like a direct toggle to this switch. prices were negative today and the switch works as you can see. 

I currently will setup a trigger to remind me when there are negative prices and then i manually toggle. 
The inverter model is: GW5K-DT

[image]
[image]

DrBlokmeister, 2024-05-12:
Thanks for sharing! This is indeed the switch I was talking about.

TomEkk33, 2024-05-18:
> @DrBlokmeister 
> You can hear the internal relays click inside the inverter when you press the button. 

Well DRED exactly does this in ET inverter. All is stopped PV=0W, Bat=0W Grid=~0W
DRED toggle does restart of inverter. When DRED is unchecked (inactive) inverter returns to normal 
work within next ~90 seconds.

It is useful to remotely restart inverter when it doesn't work as expected. By default it is inactive in HA
[image]

mletenay, 2024-05-25:
@DrBlokmeister @kokkie-d The screenshot helped, on DT (grid-only) inverters it is controlled by 2 modbus registers 40330 (Power On) and 40331 (Power Off).
I will try to implement it soon.

mletenay, 2024-05-26:
Done in release v0.9.9.26.
Enjoy.

closed 2024-05-26 by mletenay

kokkie-d, 2024-05-31:
I had a chance too test it today and it works like a charm. thank you so much.

DrBlokmeister, 2024-06-02:
Haven't tested it yet, but I trust Kokkie. Thanks for the help! I really appreciate it! :)