mletenay/home-assistant-goodwe-inverter #270: Swiching Modes  General - Eco discharge mode:  timetabels lost

Triggerfinger123, 2024-04-12:
I want to switch between the modes
- General
- Eco discharge mode
- Eco charge mode

The Eco charge & discharge do not create the "Battery work mode group".  
When creating it in the solargo app, it works.
Switching to General mode, they are deleted. 

I have a GW 20K-ET inverter.  Can anyone help me with this problem?

mletenay, 2024-04-12:
This is intentional.
If you want to keep the schedule specified in inverter, use standard "Eco Mode".
Eco charge/Eco discharge modes are special "synthetic" modes, which set the schedule 1 to "always" and clear the rest/
HA is then assumed to use automations to switch it on/off when required.

Triggerfinger123, 2024-04-13:
Switching from "Eco Mode" to "Eco Charge/Discharge Mode" the schedule 1 is set active. => OK
Switching from ""Eco Charge/Discharge Mode" to "Eco Mode" it is not deactivated!  => NOK

saltydog256, 2024-09-22:
And presumably set the parameters (like % power) or use the appropriate sliders?  Slick implementation, I have a 10KET, SBP5000, 5038EM and 3048EM all using it on one installation with 40kWh of storage in 3 different locations.
Perhaps like Triggerfinger, sometimes the changeover is not consistent, but I'll post if I nail it.  There are a lot of moving parts.