mletenay/home-assistant-goodwe-inverter #271: Feature request: Operation mode 9 and 10

mmmjank, 2024-04-17:
I was wondering if there are any plans to add support for the Regional Energy Management mode (Operation mode 9 and 10) to the integration. I checked the Goodwe Modbus Protocol documentation, and it seems like it could be a valuable addition.

[image]

mletenay, 2024-04-29:
That's a first time I see this detailed protocol documentation.
Would be nice if you can pass me the docs (e.g. via private message on discord).
It seems the work modes will need some re-design, but should be doable.

sunnysingh-84, 2024-05-04:
I wonder if these Modbus controls are specific to the EMS port only (usually a RJ45 port shared with the Parallel Inverter port), or can these be accessed via all method of GoodWe modbus access? (Modbus RTU , TCP etc..)

mmmjank, 2024-06-13:
> That's a first time I see this detailed protocol documentation. Would be nice if you can pass me the docs (e.g. via private message on discord). It seems the work modes will need some re-design, but should be doable.

I'll send you the documentation I found on the Goodwe support site via discord.

rinners, 2024-08-21:
> That's a first time I see this detailed protocol documentation. Would be nice if you can pass me the docs (e.g. via private message on discord). It seems the work modes will need some re-design, but should be doable.

Yesterday I updated DSP firmware from 7 to 8 on my GW6000 and the work mode selector is no longer available in home assistant.