mletenay/home-assistant-goodwe-inverter #272: ES5048 export limit settings with 9.9.9.17

robotz0mbie, 2024-04-21:
After updating to the latest version, 9.9.9.17, I noticed the setting for export power limit had changed to % (from an amount in W), as my automations were throwing up a bunch of errors. I converted all my W values to % of 5000 (I have 5048D_ES). Should be right? 
The slider for Export power limit can now only go up to 100. Ok...!
However... the inverter now behaves as if it is getting sent an export limit of 0-100W ... not percent... :O so even at 100% set in HA via the slider or automation, it only exports 50-100W (it tends to undershoot the export limit by a little).

Not sure what the problem there is, also not mentioned in the changelog? I rolled back to .16 and everything is OK again (back to W, not %). I have not tried rebooting the inverter or anything, but anyone else seen this bug?

thx

rans100, 2024-04-22:
I have exactly the same problem: export limits in automations in my case switched between 50W (almost no export)  and 10000W (full export / no limit), all my  automations failed. This because from version 0.9.9.17 on, percentage was used as export limit instead of watts. When I tested this, I set my export limit at 100% in HA but it shows 100W export limit in the goodwe app, so this is not implemented correctly.
I also had to roll back to v0.9.9.16

I'm not able to fix this myself but if someone needs testing a fix, let me know!

Goodwe inverter ES5048
HA 2024.4.3

robotz0mbie, 2024-04-22:
good to see its not just me :) on the plus side, 9.9.9.17 was the first version where the Eco mode SOC actually worked for me (previously had no effect), so thats a big improvment... the export issue is a bit of a problem tho heh.

mletenay, 2024-04-23:
Sorry guys, something must went wrong there.
The % vs W is there for ages (the DT family is using %).
Please revert to 0.9.9.16 until I figure out what went wrong ...

mletenay, 2024-05-05:
Fixed in v0.9.9.18

closed 2024-05-05 by mletenay