mletenay/home-assistant-goodwe-inverter #273: Can't fully charge my batteries

arkovda, 2024-04-24:
Hi And thank you for good work.
I can't charge my batteries more that 92% even doe I have 7-10kW from the sun
This has started the last weeks or so.
from 90(I guess) to 93 it turns on and off.
GW10K-ET

mletenay, 2024-04-28:
This is quite normal, resp. I observe it on my 10K-ET too, it rarely reports battery charge greater then 97-98% most often it is 93-95%. 
It is not fault of the HA integration, it just takes value reported by inverter and it's the inverter and battery management which takes care of it.
When I was investigating, at least on my pylontech batteries, it seems inverter reports lowest charge value of the battery modules.
I had 5, now 8 battery modules and their charge varies a bit, some are charged to 99% some to lower values and it is the value of the weakest module reported from BMS back to inverter. And it is not always the same module which is weakest. At least that is my observation.

In any case, there is nothing we can do about it and it is "normal" behavior.

closed 2024-04-28 by mletenay

arkovda, 2024-04-29:
Thank you for reply