mletenay/home-assistant-goodwe-inverter #280: Not working since v0.9.9.18

kolossboss, 2024-05-10:
Hi,

for me, the integration is not working since [v0.9.9.18]. 
I got either no data and the entities shows `unavailable` and the Integration is initializing or the entities shows a value but did not update for hours.
If I went back to v0.9.9.17 everything went normal again.
I'm running an **GW3648-EM Inverter**.

Not sure, how I help solving this issue.

Thats the only Log entry:

`Error fetching GoodWe data: ('No valid response received even after 6 retries', 4)`

Legnava89, 2024-05-10:
I have the same problem. The entity sensor.today_s_pv_generation and sensor.today_battery_charge are not updated always after midnight.
if I go back to version v0.9.9.17, everything is fine.
It doesn't work since v0.9.9.18.
I have a GW10K-ET converter

mletenay, 2024-05-10:
@Legnava89 this is probably the problem of https://github.com/mletenay/home-assistant-goodwe-inverter/discussions/279

@kolossboss Could you please set the log level to DEBUG and provide the logs ?

Legnava89, 2024-05-10:
@mletenay  Yes, it's the same problem as[ here ](https://github.com/mletenay/home-assistant-goodwe-inverter/discussions/279)

mletenay, 2024-05-13:
@kolossboss Could you please enable debug log and attach the log output ?

kolossboss, 2024-05-13:
Hi, thanks for reply. Sorry for late response. 
I can do that, this evening.I came back from holiday today.

kolossboss, 2024-05-14:
Hi, today I enabled the debug log and installed the latest version.

thats the output.

```
Home Assistant Core
2024-05-14 21:04:04.106 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration goodwe which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-14 21:04:16.712 DEBUG (MainThread) [goodwe] Connecting to ES family inverter at 192.168.2.114:8899.
2024-05-14 21:04:16.712 DEBUG (MainThread) [goodwe.protocol] Creating lock instance for current event loop.
2024-05-14 21:04:16.713 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241
2024-05-14 21:04:16.904 DEBUG (MainThread) [goodwe.protocol] Response has unexpected length: 77, expected 86.
2024-05-14 21:04:16.904 DEBUG (MainThread) [goodwe.protocol] Received invalid response: aa557fc001824d313131314a4757333634382d454d20233130000000000000000000000000003933363438454d553139435830303034333630303431302d30343030302d3131023431302d3032
2024-05-14 21:04:16.904 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #1/6
2024-05-14 21:04:17.320 DEBUG (MainThread) [goodwe.protocol] Response has unexpected length: 16, expected 86.
2024-05-14 21:04:17.320 DEBUG (MainThread) [goodwe.protocol] Received invalid response: aa557fc001824d313131314a47573336
2024-05-14 21:04:17.320 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #2/6
2024-05-14 21:04:17.745 DEBUG (MainThread) [goodwe.protocol] Response has unexpected length: 82, expected 86.
2024-05-14 21:04:17.745 DEBUG (MainThread) [goodwe.protocol] Received invalid response: aa557fc001824d313131314a4757333634382d454d20233130000000000000000000000000003933363438454d553139435830303034333630303431302d30343030302d3131023431302d30323033342d31
2024-05-14 21:04:17.745 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #3/6
2024-05-14 21:04:18.161 DEBUG (MainThread) [goodwe.protocol] Response has unexpected length: 32, expected 86.
2024-05-14 21:04:18.161 DEBUG (MainThread) [goodwe.protocol] Received invalid response: aa557fc001824d313131314a4757333634382d454d2023313000000000000000
2024-05-14 21:04:18.162 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #4/6
2024-05-14 21:04:18.584 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001824d313131314a4757333634382d454d20233130000000000000000000000000003933363438454d553139435830303034333630303431302d30343030302d3131023431302d30323033342d313918102a
2024-05-14 21:04:18.585 DEBUG (MainThread) [goodwe] Connected to inverter GW3648-EM, S/N:93648EMU19CX0004.
2024-05-14 21:04:18.585 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2024-05-14 21:04:18.819 DEBUG (MainThread) [goodwe.protocol] Response has unexpected length: 74, expected 149.
2024-05-14 21:04:18.819 DEBUG (MainThread) [goodwe.protocol] Received invalid response: aa557fc001868c052500000003040001000211000300500110006f005f00cd00005300536402000001091b001a0002138201091b0001025a138201020173000000000001292c00005423
2024-05-14 21:04:18.819 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245 - retry #1/6
2024-05-14 21:04:19.124 DEBUG (MainThread) [goodwe.protocol] Response has unexpected length: 76, expected 149.
2024-05-14 21:04:19.124 DEBUG (MainThread) [goodwe.protocol] Received invalid response: aa557fc001868c09ac00000202cc0001000211000300500110006e005f00cd00005300536402000001091b00190002138301091b00010249138301020173000000000001292c0000542300fc
2024-05-14 21:04:19.125 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245 - retry #2/6
2024-05-14 21:04:19.600 DEBUG (MainThread) [goodwe.protocol] Response has unexpected length: 31, expected 149.
2024-05-14 21:04:19.600 DEBUG (MainThread) [goodwe.protocol] Received invalid response: aa557fc001868c052b00000002f40001000211000300500110006d005f00cd
2024-05-14 21:04:19.600 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245 - retry #3/6
2024-05-14 21:04:20.080 DEBUG (MainThread) [goodwe.protocol] Response has unexpected length: 21, expected 149.
2024-05-14 21:04:20.080 DEBUG (MainThread) [goodwe.protocol] Received invalid response: aa557fc001868c0628000002030a00010002110003
2024-05-14 21:04:20.080 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245 - retry #4/6
2024-05-14 21:04:20.556 DEBUG (MainThread) [goodwe.protocol] Response has unexpected length: 10, expected 149.
2024-05-14 21:04:20.557 DEBUG (MainThread) [goodwe.protocol] Received invalid response: aa557fc001868c06de00
2024-05-14 21:04:20.558 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245 - retry #5/6
2024-05-14 21:04:21.046 DEBUG (MainThread) [goodwe.protocol] Response has unexpected length: 117, expected 149.
2024-05-14 21:04:21.046 DEBUG (MainThread) [goodwe.protocol] Received invalid response: aa557fc001868c0708000002000000010002110003005001100076005f00cd00005300536402000001091b001c0003138201091b0001026b138201020173000000000001292c0000542300fc00df000264fe025c01003000000000010000030303000040000018050e150318000000000000000000
2024-05-14 21:04:21.046 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245 - retry #6/6
2024-05-14 21:04:21.523 DEBUG (MainThread) [goodwe.protocol] Response has unexpected length: 71, expected 149.
[... cut ...]
        "network_retries": 6,
        "network_timeout": 30
      },
      "pref_disable_new_entities": false,
      "pref_disable_polling": false,
      "source": "user",
      "unique_id": "93648EMU19CX0004",
      "disabled_by": null
    },
    "inverter": {
      "model_name": "GW3648-EM",
      "rated_power": 0,
      "firmware": "1111J",
      "arm_firmware": null,
      "dsp1_version": 11,
      "dsp2_version": 11,
      "dsp_svn_version": null,
      "arm_version": 19,
      "arm_svn_version": null
    }
  }
}
```

mletenay, 2024-05-15:
Hmm, that is odd. It seems you are receiving truncated/incomplete responses.
Are you sure 0.9.9.17 works properly ? Can you try again ?

kolossboss, 2024-05-15:
Yes, 
I went back to v0.9.9.17 and everything works fine again.

mletenay, 2024-05-15:
There were changes in the code, but the log files look like problem is on network layer, not application.
I do not have explanation for it (yet).
One minor thing, you have timeout value 30 and scan rate 5 - it makes little sense - timeout should be lower than scan rate.

kolossboss, 2024-05-15:
I always have problems with the network connection. All Goodwe entities, shows sometimes unavailable for a short time. In the past I played a lot with these retry and timeout settings and scan rate 5 and timeout 30 works ok^^
Now I set to scan rate 15 and timeout 10 and will test.
I can also test this with the newer version.
And yes. I tried v0.9.9.18 and 19.
Both same issue.

kolossboss, 2024-05-16:
Ok I tried again v0.9.9.19 with the new scan rate and timeout settings.
But still not working. All entities are unavailable.
I went back to 17 and it starts working again

mletenay, 2024-05-16:
That's unfortunate, since I do not have an explanation for that.
What HW are you running on ?
Can you try to test just plain python scripts outside HA ?

kolossboss, 2024-05-16:
The hardware, HA is running on?
A simple Intel Celeron Mini PC with N5105 CPU and Home Assistant OS.
My network hardware is completely unifi devices, and I have no other network issues at all.

Yes, I think I can run a Python script.

mletenay, 2024-05-16:
Ok, let's try to isolate the problem.
Install pyhton3 (preferably on other machine than HA), install git.
Checkout the library - git clone https://github.com/marcelblijleven/goodwe.git
Then try to run inverter_check.py from tests folder - adjusting the IP address and family to 'ES'

kolossboss, 2024-05-16:
Hi,
I run the script. I adjusted the family to EM because I have an GW3648-EM Inverter. Hope this is correct.

Thats the output:

```
2024-05-16 20:38:22,219 __init__(59) - DEBUG: Using selector: KqueueSelector
2024-05-16 20:38:22,219 connect(55) - DEBUG: Connecting to EM family inverter at 192.168.2.114:8899.
2024-05-16 20:38:22,219 _ensure_lock(56) - DEBUG: Creating lock instance for current event loop.
2024-05-16 20:38:22,220 _send_request(163) - DEBUG: Sending: aa55c07f0102000241
2024-05-16 20:38:22,538 _validate_response(465) - DEBUG: Response has unexpected length: 11, expected 86.
2024-05-16 20:38:22,538 datagram_received(130) - DEBUG: Received invalid response: aa557fc001824d31313131
2024-05-16 20:38:22,538 _send_request(161) - DEBUG: Sending: aa55c07f0102000241 - retry #1/3
2024-05-16 20:38:22,965 _validate_response(465) - DEBUG: Response has unexpected length: 69, expected 86.
2024-05-16 20:38:22,965 datagram_received(130) - DEBUG: Received invalid response: aa557fc001824d313131314a4757333634382d454d20233130000000000000000000000000003933363438454d553139435830303034333630303431302d30343030302d31
2024-05-16 20:38:22,966 _send_request(161) - DEBUG: Sending: aa55c07f0102000241 - retry #2/3
2024-05-16 20:38:23,384 datagram_received(127) - DEBUG: Received: aa557fc001824d313131314a4757333634382d454d20233130000000000000000000000000003933363438454d553139435830303034333630303431302d30343030302d3131023431302d30323033342d313918102a
2024-05-16 20:38:23,385 connect(57) - DEBUG: Connected to inverter GW3648-EM, S/N:93648EMU19CX0004.
Identified inverter
- Model: GW3648-EM
- SerialNr: 93648EMU19CX0004
- Rated power: 0
- A/C output type: None
- Firmware: 1111J
- ARM firmware: None
- Modbus version: None
- DSP1 version: 11
- DSP2 version: 11
- DSP svn version: None
- Arm version: 19
- ARM svn version: None

2024-05-16 20:38:23,386 __init__(59) - DEBUG: Using selector: KqueueSelector
2024-05-16 20:38:23,386 _ensure_lock(56) - DEBUG: Creating lock instance for current event loop.
2024-05-16 20:38:23,386 close_transport(186) - DEBUG: Failed to close transport.
2024-05-16 20:38:23,387 _send_request(163) - DEBUG: Sending: aa55c07f0106000245
2024-05-16 20:38:23,861 _validate_response(465) - DEBUG: Response has unexpected length: 34, expected 149.
2024-05-16 20:38:23,861 datagram_received(130) - DEBUG: Received invalid response: aa557fc001868c04a600010002f4000200020e00030050011801c2005f00cd000056
2024-05-16 20:38:23,862 _send_request(161) - DEBUG: Sending: aa55c07f0106000245 - retry #1/3
2024-05-16 20:38:24,347 _validate_response(465) - DEBUG: Response has unexpected length: 108, expected 149.
2024-05-16 20:38:24,348 datagram_received(130) - DEBUG: Received invalid response: aa557fc001868c02b500010003e6000200020e00030050011801c2005f00cd00005600566402000001092f005a0004138601092f000208691385010201b50000000000012b090000545300ba00710002662a088e0100300000000001000003be030000400000180510142518
2024-05-16 20:38:24,348 _send_request(161) - DEBUG: Sending: aa55c07f0106000245 - retry #2/3
2024-05-16 20:38:24,627 _validate_response(465) - DEBUG: Response has unexpected length: 97, expected 149.
2024-05-16 20:38:24,627 datagram_received(130) - DEBUG: Received invalid response: aa557fc001868c029100010003fb000200020e00030050011801c1005f00cd000056005664020000010930005900041385010930000208731385010201b50000000000012b090000545300ba00710002662a08870100300000000001000003bf03
2024-05-16 20:38:24,627 _send_request(161) - DEBUG: Sending: aa55c07f0106000245 - retry #3/3
2024-05-16 20:38:24,995 _validate_response(465) - DEBUG: Response has unexpected length: 50, expected 149.
2024-05-16 20:38:24,996 datagram_received(130) - DEBUG: Received invalid response: aa557fc001868c04ab00010002e0000200020e00030050011801be005f00cd00005600566402000001092c005a0004138501
2024-05-16 20:38:24,996 _retry_mechanism(177) - DEBUG: Max number of retries (3) reached, request aa55c07f0106000245 failed.
2024-05-16 20:38:24,996 connection_lost(117) - DEBUG: Socket closed.
Traceback (most recent call last):
  File "/Users/macminiserver/test/goodwe/tests/inverter_check.py", line 62, in <module>
    response = asyncio.run(inverter.read_runtime_data())
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/Users/macminiserver/test/goodwe/tests/../../goodwe/goodwe/es.py", line 206, in read_runtime_data
    response = await self._read_from_socket(self._READ_DEVICE_RUNNING_DATA)
  File "/Users/macminiserver/test/goodwe/tests/../../goodwe/goodwe/inverter.py", line 128, in _read_from_socket
    raise RequestFailedException(f'No valid response received even after {self._protocol.retries} retries',
goodwe.exceptions.RequestFailedException: ('No valid response received even after 3 retries', 1)

```

kolossboss, 2024-05-17:
Hi, 
Thanks for the update, but its still not working.
Now, the whole integration is not loading correct.

Thats the log.

```
Logger: homeassistant.config_entries
Quelle: config_entries.py:575
Erstmals aufgetreten: 06:12:32 (1 Vorkommnisse)
Zuletzt protokolliert: 06:12:32

Error setting up entry GoodWe for goodwe
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 575, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/goodwe/__init__.py", line 31, in async_setup_entry
    protocol = entry.options.get(CONF_PROTOCOL, entry.data[CONF_PROTOCOL])
                                                ~~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'protocol'
```

kolossboss, 2024-05-17:
I installed v0.9.9.21. 
But there is still something wrong.

- The integration is now called without "experimental" in der Integration tab. 
- It still show v0.9.9.20 in the Integration tab
- Entities still not available or not updating
- Cant click on configuration (screenshot)

Log:

```
Home Assistant Core
2024-05-17 08:41:18.101 WARNING (SyncWorker_6) [homeassistant.loader] We found a custom integration goodwe which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 130, in datagram_received
File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 468, in <lambda>
File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 488, in _validate_response
File "/config/custom_components/goodwe/config_flow.py", line 69, in async_step_init
File "/config/custom_components/goodwe/config_flow.py", line 69, in async_step_init
2024-05-17 08:47:06.376 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% if is_state('sensor.goodwe_battery_mode', '3') %}
{{ states('sensor.goodwe_pbattery1')|float * -1}}
{% endif %}' but no default was specified') while processing template 'Template<template=({% if is_state('sensor.goodwe_battery_mode', '3') %}
{{ states('sensor.goodwe_pbattery1')|float * -1}}
{% endif %}) renders=6>' for attribute '_attr_native_value' in entity 'sensor.goodwe_battery_charge'
2024-05-17 08:47:06.380 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: int got invalid input 'unavailable' when rendering template '{% set battery_capacity = 14 %} {% set battery_current = (battery_capacity * (states('sensor.goodwe_battery_soc')|int/100))|round(2) %} {{ battery_current }}' but no default was specified') while processing template 'Template<template=({% set battery_capacity = 14 %} {% set battery_current = (battery_capacity * (states('sensor.goodwe_battery_soc')|int/100))|round(2) %} {{ battery_current }}) renders=6>' for attribute '_attr_native_value' in entity 'sensor.battery_available'
2024-05-17 08:47:06.382 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: int got invalid input 'unavailable' when rendering template '{% set battery_capacity = 14 %} {% set battery_current = (battery_capacity * (states('sensor.goodwe_battery_soc')|int/100))|round(2) %} {% set battery_to_charge = (battery_capacity - battery_current)|round(2) %} {{ battery_to_charge }}' but no default was specified') while processing template 'Template<template=({% set battery_capacity = 14 %} {% set battery_current = (battery_capacity * (states('sensor.goodwe_battery_soc')|int/100))|round(2) %} {% set battery_to_charge = (battery_capacity - battery_current)|round(2) %} {{ battery_to_charge }}) renders=6>' for attribute '_attr_native_value' in entity 'sensor.battery_tocharge'
2024-05-17 08:47:06.385 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% if states('sensor.goodwe_pgrid')|float < 0 %}
{{ (states('sensor.goodwe_pgrid')|float * -1)|round(0) }}
{% endif %}' but no default was specified') while processing template 'Template<template=({% if states('sensor.goodwe_pgrid')|float < 0 %}
{{ (states('sensor.goodwe_pgrid')|float * -1)|round(0) }}
2024-05-17 08:47:06.387 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% if states('sensor.goodwe_pgrid')|float > 0 %}
{{ states('sensor.goodwe_pgrid')|float }}
{% endif %}' but no default was specified') while processing template 'Template<template=({% if states('sensor.goodwe_pgrid')|float > 0 %}
{{ states('sensor.goodwe_pgrid')|float }}
2024-05-17 08:47:06.393 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% if (states('sensor.goodwe_house_consumption')|float - states('sensor.goodwe_pback_up')|float) > 0 %}
{{ (states('sensor.goodwe_house_consumption')|float - states('sensor.goodwe_pback_up')|float)|round(0) }}
{% endif %}' but no default was specified') while processing template 'Template<template=({% if (states('sensor.goodwe_house_consumption')|float - states('sensor.goodwe_pback_up')|float) > 0 %}
{{ (states('sensor.goodwe_house_consumption')|float - states('sensor.goodwe_pback_up')|float)|round(0) }}
2024-05-17 08:47:06.395 DEBUG (MainThread) [custom_components.goodwe.coordinator] No response received (streak of 1)
2024-05-17 08:47:06.401 DEBUG (MainThread) [goodwe] Connecting to ES family inverter at 192.168.2.114:8899.
2024-05-17 08:47:06.401 DEBUG (MainThread) [goodwe.protocol] Creating lock instance for current event loop.
2024-05-17 08:47:06.404 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241
2024-05-17 08:47:06.596 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001824d313131314a4757333634382d454d20233130000000000000000000000000003933363438454d553139435830303034333630303431302d30343030302d3131023431302d30323033342d313918102a
[... cut ...]
2024-05-17 08:47:12.420 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc00189560000000000000000000000000001000100000001000a00de02300085004001bd0013000f000000000000ffff0214001e001e00811194000000000000023f00000000000000000000000001f401f401bd000b0000000c0c56
2024-05-17 08:47:12.423 DEBUG (MainThread) [goodwe.protocol] Sending: READ 6 registers from 47547 (f703b9bb00068427)
2024-05-17 08:47:12.450 DEBUG (MainThread) [goodwe.protocol] Received response fragment: aa55f7030c0000093700
2024-05-17 08:48:08.048 ERROR (MainThread) [homeassistant.components.number] Setup of platform goodwe is taking longer than 60 seconds. Startup will proceed without waiting any longer.
```
[image]

mletenay, 2024-05-17:
Aaaaaaagh.
Try v0.9.9.22 and then in config UI disable the "TCP keep alive"

kolossboss, 2024-05-17:
Sorry man...its still not working 🙃

I can now configure the Integration. I disabled TCP keep alive. I also changed Inverter family to "EM". But the family is not saving. It always goes back to "ES".

```
Home Assistant Core
2024-05-17 09:54:39.358 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration goodwe which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-17 09:54:43.568 DEBUG (MainThread) [goodwe] Connecting to ES family inverter at 192.168.2.114:8899.
2024-05-17 09:54:43.568 DEBUG (MainThread) [goodwe.protocol] Creating lock instance for current event loop.
2024-05-17 09:54:43.568 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241
2024-05-17 09:54:43.872 DEBUG (MainThread) [goodwe.protocol] Response has unexpected length: 78, expected 86.
2024-05-17 09:54:43.873 DEBUG (MainThread) [goodwe.protocol] Received invalid response: aa557fc001824d313131314a4757333634382d454d20233130000000000000000000000000003933363438454d553139435830303034333630303431302d30343030302d3131023431302d303230
2024-05-17 09:54:43.873 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #1/6
2024-05-17 09:54:44.294 DEBUG (MainThread) [goodwe.protocol] Response has unexpected length: 30, expected 86.
2024-05-17 09:54:44.294 DEBUG (MainThread) [goodwe.protocol] Received invalid response: aa557fc001824d313131314a4757333634382d454d202331300000000000
2024-05-17 09:54:44.328 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241 - retry #2/6
2024-05-17 09:54:44.720 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001824d313131314a4757333634382d454d20233130000000000000000000000000003933363438454d553139435830303034333630303431302d30343030302d3131023431302d30323033342d313918102a
2024-05-17 09:54:44.726 DEBUG (MainThread) [goodwe] Connected to inverter GW3648-EM, S/N:93648EMU19CX0004.
2024-05-17 09:54:44.726 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2024-05-17 09:54:45.281 DEBUG (MainThread) [goodwe.protocol] Response has unexpected length: 113, expected 149.
2024-05-17 09:54:45.281 DEBUG (MainThread) [goodwe.protocol] Received invalid response: aa557fc001868c0e830013020f1000100202160003005001200190005f00cd00002c002c6403000001093b001e0011138801093b0001036f1388010201ca0000000000012b4b00005460002700370002667cfc4c01003000000000010000fc5d03000800000018051109352b0000000000
2024-05-17 09:54:45.290 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245 - retry #1/6
File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 130, in datagram_received
File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 468, in <lambda>
File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 488, in _validate_response
2024-05-17 09:59:40.901 DEBUG (MainThread) [custom_components.goodwe.coordinator] No response received (streak of 1)
2024-05-17 09:59:40.902 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 296.176 seconds (success: True)
2024-05-17 09:59:40.903 DEBUG (MainThread) [goodwe.protocol] Socket closed.
2024-05-17 09:59:40.942 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 130, in datagram_received
File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 468, in <lambda>
File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 488, in _validate_response
2024-05-17 10:00:40.930 ERROR (MainThread) [homeassistant.components.number] Setup of platform goodwe is taking longer than 60 seconds. Startup will proceed without waiting any longer.
2024-05-17 10:00:40.930 ERROR (MainThread) [homeassistant.components.select] Setup of platform goodwe is taking longer than 60 seconds. Startup will proceed without waiting any longer.
2024-05-17 10:00:40.942 ERROR (MainThread) [homeassistant.components.switch] Setup of platform goodwe is taking longer than 60 seconds. Startup will proceed without waiting any longer.
```
[image]

kolossboss, 2024-05-20:
Sorry still not working with 0.9.9.23.
v0.9.9.17 still works fine.

Its not possible the change Inverter Family to `EM`. It always goes back to `ES`. Not sure if this is a problem. In the log it says `connecting to EM family`

Thats the log:

```
2024-05-20 10:06:50.675 WARNING (SyncWorker_6) [homeassistant.loader] We found a custom integration goodwe which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-05-20 10:07:02.092 DEBUG (MainThread) [goodwe] Connecting to EM family inverter at 192.168.2.114:8899.
2024-05-20 10:07:02.092 DEBUG (MainThread) [goodwe.protocol] Creating lock instance for current event loop.
2024-05-20 10:07:02.093 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0102000241
2024-05-20 10:07:02.271 DEBUG (MainThread) [goodwe.protocol] Received: aa557fc001824d313131314a4757333634382d454d20233130000000000000000000000000003933363438454d553139435830303034333630303431302d30343030302d3131023431302d30323033342d313918102a
2024-05-20 10:07:02.272 DEBUG (MainThread) [goodwe] Connected to inverter GW3648-EM, S/N:93648EMU19CX0004.
2024-05-20 10:07:02.272 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0106000245
2024-05-20 10:07:02.756 DEBUG (MainThread) [goodwe.protocol] Received response fragment: aa557fc001868c0e72000b020ea4000b0202130003005001050044005f00cd0000280028640300000109280011000413870109280001017213860102019e0000000000012d54000054a80001002700
2024-05-20 10:11:52.403 DEBUG (MainThread) [custom_components.goodwe.coordinator] No response received (streak of 1)
2024-05-20 10:11:52.403 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 290.131 seconds (success: True)
2024-05-20 10:11:52.437 DEBUG (MainThread) [goodwe.protocol] Sending: aa55c07f0109000248
2024-05-20 10:11:52.619 DEBUG (MainThread) [goodwe.protocol] Received another response fragment: aa557fc00189560000000000000000000000000001000100000001000a00de02300085004001bd0013000f000000000000ffff0214001e001e00811194000000000000023f00000000000000000000000001f401f401bd000b0000000c.
2024-05-20 10:11:52.619 DEBUG (MainThread) [goodwe.protocol] Received response fragment: aa557fc001868c0e72000b020ea4000b0202130003005001050044005f00cd0000280028640300000109280011000413870109280001017213860102019e0000000000012d54000054a80001002700aa557fc00189560000000000000000000000000001000100000001000a00de02300085004001bd0013000f000000000000ffff0214001e001e00811194000000000000023f00000000000000000000000001f401f401bd000b0000000c
2024-05-20 10:12:52.438 ERROR (MainThread) [homeassistant.components.number] Setup of platform goodwe is taking longer than 60 seconds. Startup will proceed without waiting any longer.
2024-05-20 10:12:52.439 ERROR (MainThread) [homeassistant.components.select] Setup of platform goodwe is taking longer than 60 seconds. Startup will proceed without waiting any longer.
2024-05-20 10:12:52.454 ERROR (MainThread) [homeassistant.components.switch] Setup of platform goodwe is taking longer than 60 seconds. Startup will proceed without waiting any longer.
```
[image]

mletenay, 2024-05-20:
ES and EM is the same stuff, so not issue.
But I'm running out of ideas  :-(

jiriknez, 2024-05-20:
From version 0.9.9.18 I have problem with 10ET, that after 15-30 minutes integration terminates with this error. Every 30 minutes I have to reload config or restart HA.

Error doing job: Exception in callback _SelectorDatagramTransport._read_ready()
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 131, in datagram_received
    if self.command.validator(data):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 579, in <lambda>
    lambda x: validate_modbus_rtu_response(x, cmd, offset, value),
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/goodwe/modbus.py", line 206, in validate_modbus_rtu_response
    raise RequestRejectedException(failure_code)
goodwe.exceptions.RequestRejectedException: ILLEGAL DATA ADDRESS

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 1238, in _read_ready
    self._protocol.datagram_received(data, addr)
  File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 149, in datagram_received
    self.response_future.set_exception(ex)
asyncio.exceptions.InvalidStateError: invalid state

[home-assistant_goodwe_2024-05-20T09-25-48.662Z.log](https://github.com/mletenay/home-assistant-goodwe-inverter/files/15377552/home-assistant_goodwe_2024-05-20T09-25-48.662Z.log)

kolossboss, 2024-05-20:
> ES and EM is the same stuff, so not issue.
> 
> But I'm running out of ideas  :-(

If there is any way I can help you, feel free to ask.
I can test all new versions, if I can always downgrade to v0.9.9.17

I really need this integration.😬

dl6dbh, 2024-05-20:
i have tree gw5048es . i have the same poblem

mletenay, 2024-05-20:
@jiriknez Your issue seems to be of different kind. 
Please try the fresh release and if the problem persists, file a separate ticket so I can track it properly.

mletenay, 2024-05-20:
@kolossboss I cannot find anything in the code which would explain this behavior.
At this moment I see only one way - you should checkout the git repo and start reverting back to the last commit which will work for you.
Then I can look at it once again and try to figure out what the h***...

kolossboss, 2024-05-21:
I'm not exactly sure what you mean. I'm not so familiar with github.
Last working Version is v0.9.9.17. 
In v0.9.9.18 is only one commit.

Also,
In the release notes you describes a "major Rewrite of the communication protocol" 

I have no clue what to do...

mletenay, 2024-05-21:
I will give you exact instructions, don't worry.
But I guess using discord channel will be more efficient that this github issue.

JoseMadbird, 2024-05-21:
How to downgrade i will to test, as still not working for me....

mletenay, 2024-05-21:
1) Assuming python >= 3.8 is already installed
2) Install GIT - https://git-scm.com/downloads
3) Clone the repo by executing "git clone https://github.com/marcelblijleven/goodwe.git"
4) Go to just cloned goodwe folder and edit the file "tests\inverter_check.py".
5) Set the IP_ADDRESS to your inverter IP and change the FAMILY to "ES"
6) Execute "python tests\inverter_check.py"
7) Observe behavior of the latest code, try couple of times.
8) If not OK, then move back in history to start of May by executing "git reset --hard 32d9422"
9) Test again, if still not OK, then move back one more commit by executing "git reset --hard HEAD~1"
10) Repeat step 9 until the the codes behaves properly again OK
11) Execute "git log -1" and see the commit ID and message of the last known good moment.

kolossboss, 2024-05-22:
Hi,
thanks for the instructions.

I cloned the repo and executed the inverter_check.py multiple times.

Most of the time there is "no response" and that looks like this. (First log)

But sometimes `(1 of 10)` I get some sensor information from the inverter. (second log)

Then I startet the `git reset --hard HEAD~1` thing. I'm now at `71c79a9 `, but still not working. OR I dont know how a good response looks like?
I expected a lots of sensor informations but I only get the inverter informations. But the Inverter informations I get on any versions. 
(third Log)
Should I go even further?

How can I contact you on Discord??

```
macminiserver@macminiserver tests % python3 inverter_check.py
2024-05-22 21:02:33,788 __init__(59) - DEBUG: Using selector: KqueueSelector
2024-05-22 21:02:33,788 connect(52) - DEBUG: Connecting to ES family inverter at 192.168.2.114:8899.
2024-05-22 21:02:33,788 _ensure_lock(59) - DEBUG: Creating lock instance for current event loop.
2024-05-22 21:02:33,788 _send_request(179) - DEBUG: Sending: aa55c07f0102000241
2024-05-22 21:02:34,082 datagram_received(141) - DEBUG: Received response fragment (67 of 86): aa557fc001824d313131314a4757333634382d454d20233130000000000000000000000000003933363438454d553139435830303034333630303431302d3034303030
2024-05-22 21:02:35,082 _retry_mechanism(189) - DEBUG: Failed to receive response to aa55c07f0102000241 in time (1s).
2024-05-22 21:02:35,083 _send_request(177) - DEBUG: Sending: aa55c07f0102000241 - retry #1/3
2024-05-22 21:02:35,336 datagram_received(141) - DEBUG: Received response fragment (55 of 86): aa557fc001824d313131314a4757333634382d454d20233130000000000000000000000000003933363438454d55313943583030303433
2024-05-22 21:02:36,337 _retry_mechanism(189) - DEBUG: Failed to receive response to aa55c07f0102000241 in time (1s).
2024-05-22 21:02:36,337 _send_request(177) - DEBUG: Sending: aa55c07f0102000241 - retry #2/3
2024-05-22 21:02:36,687 datagram_received(141) - DEBUG: Received response fragment (85 of 86): aa557fc001824d313131314a4757333634382d454d20233130000000000000000000000000003933363438454d553139435830303034333630303431302d30343030302d3131023431302d30323033342d31391810
2024-05-22 21:02:37,688 _retry_mechanism(189) - DEBUG: Failed to receive response to aa55c07f0102000241 in time (1s).
2024-05-22 21:02:37,689 _send_request(177) - DEBUG: Sending: aa55c07f0102000241 - retry #3/3
2024-05-22 21:02:38,690 _retry_mechanism(193) - DEBUG: Max number of retries (3) reached, request aa55c07f0102000241 failed.
2024-05-22 21:02:38,691 connection_lost(120) - DEBUG: Socket closed.
Traceback (most recent call last):
  File "/Users/macminiserver/test/goodwe/tests/inverter_check.py", line 43, in <module>
    inverter = asyncio.run(goodwe.connect(IP_ADDRESS, PORT, FAMILY, COMM_ADDR, TIMEOUT, RETRIES))
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/Users/macminiserver/test/goodwe/tests/../../goodwe/goodwe/__init__.py", line 53, in connect
    await inv.read_device_info()
  File "/Users/macminiserver/test/goodwe/tests/../../goodwe/goodwe/es.py", line 186, in read_device_info
    response = await self._read_from_socket(self._READ_DEVICE_VERSION_INFO)
  File "/Users/macminiserver/test/goodwe/tests/../../goodwe/goodwe/inverter.py", line 127, in _read_from_socket
    raise RequestFailedException(f'No valid response received even after {self._protocol.retries} retries',
goodwe.exceptions.RequestFailedException: ('No valid response received even after 3 retries', 1)
macminiserver@macminiserver tests % python3 inverter_check.py
2024-05-22 21:03:54,715 __init__(59) - DEBUG: Using selector: KqueueSelector
2024-05-22 21:03:54,715 connect(52) - DEBUG: Connecting to ES family inverter at 192.168.2.114:8899.
2024-05-22 21:03:54,715 _ensure_lock(59) - DEBUG: Creating lock instance for current event loop.
2024-05-22 21:03:54,715 _send_request(179) - DEBUG: Sending: aa55c07f0102000241
2024-05-22 21:03:54,755 datagram_received(141) - DEBUG: Received response fragment (77 of 149): aa557fc001868c078500000207bc00010202010003005001000005005f00cd0000130013640100000109140001022f13870109140001022f1387010201270000000000012f02000054e3003f00
2024-05-22 21:03:55,169 datagram_received(141) - DEBUG: Received response fragment (16 of 86): aa557fc001824d313131314a47573336
2024-05-22 21:03:56,170 _retry_mechanism(189) - DEBUG: Failed to receive response to aa55c07f0102000241 in time (1s).
2024-05-22 21:03:56,171 _send_request(177) - DEBUG: Sending: aa55c07f0102000241 - retry #1/3
2024-05-22 21:03:56,359 datagram_received(135) - DEBUG: Received: aa557fc001824d313131314a4757333634382d454d20233130000000000000000000000000003933363438454d553139435830303034333630303431302d30343030302d3131023431302d30323033342d313918102a
2024-05-22 21:03:56,359 connect(54) - DEBUG: Connected to inverter GW3648-EM, S/N:93648EMU19CX0004.
Identified inverter
- Model: GW3648-EM
- SerialNr: 93648EMU19CX0004
- Rated power: 0
- A/C output type: None
- Firmware: 1111J
- ARM firmware: None
- Modbus version: None
- DSP1 version: 11
- DSP2 version: 11
- DSP svn version: None
- Arm version: 19
[... 126 lines cut ...]

```
macminiserver@macminiserver tests % git reset --hard HEAD~1  
[... cut ...]
2024-05-22 21:39:06,400 datagram_received(50) - DEBUG: Received: aa557fc001824d313131314a4757333634382d454d20233130000000000000000000000000003933363438454d553139435830303034333630303431302d30343030302d3131023431302d30323033342d313918102a
2024-05-22 21:39:06,401 connect(57) - DEBUG: Connected to inverter GW3648-EM, S/N:93648EMU19CX0004.
Identified inverter
- Model: GW3648-EM
- SerialNr: 93648EMU19CX0004
- Rated power: 0
- A/C output type: None
- Firmware: 1111J
- ARM firmware: None
- Modbus version: None
- DSP1 version: 11
- DSP2 version: 11
- DSP svn version: None
- Arm version: 19
- ARM svn version: None
```

kolossboss, 2024-05-24:
No, I thing it get more stable on earlier commits.
I waited for some sensor data like the first log.

But now I will look on the lengths responses.
I do it once again.

mletenay, 2024-05-24:
Data are not really relevant for the test, you can even comment out the print() command there.
What we need to observe is ratio of good and bad (incomplete, fragment or timed out) responses.

GFKSolar, 2024-05-25:
actually i have the same problem :) i just downgraded i dont really care at the moment , nothing i really need in that update i am running on a HA green

mletenay, 2024-05-25:
Yeah, but the code which is in HACS will one day get into core HA, so will have the problem then until it gets fixed ;-)

kolossboss, 2024-05-26:
@mletenay 
ok, today I looked at may terminal out from 22.05. again.
Its really hard to tell when it gets better.

Maybe at head `573a64b` I get a better response.

I also added the whole file. Maybe you can finde something.

[Terminal_output_goodwe.txt](https://github.com/mletenay/home-assistant-goodwe-inverter/files/15448588/Terminal_output_goodwe.txt)

mletenay, 2024-05-26:
Hmm, I'm afraid I do not see any relevant correlation there.
The incomplete packets seem to be completely random, no commit/version seems to stand out.
I have reviewed the code changes once again and I do not see anything which could explain change of behavior in this aspect.

You receive broken/truncated UDP packets and I have no idea why.
My guts would blame network infrastructure, but I don't have any rational explanation for that.

kolossboss, 2024-05-27:
Does it makes any sense to go down any further with the commit?

Its hard to believe that's an only network issue, because v0.9.9.17 is working just fine :-)

Today I updated to v0.9.9.26. It takes a long time until the integration is initialized, but then I get partial values. But the update rate is 3-5 minutes, and sometimes the entities are unavailable for several minutes.

Is there anything else I can do to help you?

[image]

mletenay, 2024-05-27:
I see but I do not understand why.
Did you try both checked/unchecked "Keep Alive" toggle ?

kolossboss, 2024-05-31:
Sorry for late response.

Yes, I tried all settings. 
I also thought "Keep Alive" is only for the TCP Protocol.

mletenay, 2024-05-31:
Yes, it's named that way and it is meant mostly for TCP.
But there is also a subtle change of behavior for UDP.
Could you please retry the v0.9.9.26 with both keep-alive on and off, since there was a bug there in some prev releases.

kolossboss, 2024-05-31:
I tried with Keep Alive on.
Also not working. Same behaviour.

mletenay, 2024-05-31:
Just for the record, KeepAlive ON is new behavior, with KeepAlive OFF it should behave (for UDP) the same way as in 0.9.9.17.

smileyman-1, 2024-06-10:
@kolossboss
I have the same inverter (GW3648-EM). The current versions, e.g. 0.9.9.26, work perfectly for me.
I have connected the original WLAN stick (version 1). WLAN is provided by an AVM FritzBox.
Do you have the latest firmware on the inverter and WLAN stick?

kolossboss, 2024-06-10:
Hi,

Since the beginning, I have had connection reliability problems with my inverter (losing WiFi connection indicated by a blinking LED, SolarGo sometimes not working, etc.). So, I contacted Goodwe Support, and they installed the LAN module instead.

It has gotten much better, but it’s still not perfect. I thought these reliability problems were by design and not a real issue, because the data is sent to the SEMS portal just fine.

Maybe there is a general problem with my inverter... but why does v0.9.9.17 work well enough?

In the web interface, the firmware shows V1.0.14.
In SolarGo, the firmware shows V111J.
In the SEMS portal, the firmware shows 3111119.

smileyman-1, 2024-06-10:
The WiFi module is running very stably for me. Despite being polled every second by Home Assistant, SEMS is also being filled with data. Sometimes the WiFi LED flashes and reports that it has no connection to SEMS. However, this only happens very rarely and resolves itself. I cannot see any failures in SEMS, and Home Assistant is still receiving current data without any problems.

The inverter firmware does not seem to be up to date in my opinion. I have a 1314J (which is also on Goodwe's GoogleDrive).

1.0.14 is only the version of the web interface and not the module firmware. I don't currently know the version for the LAN stick. For the WiFi module it is 1.1.3.10.

kolossboss, 2024-06-10:
Hi,
thanks for reply.
Where can I find the LAN stick version?
And how can I upgrade the Inverter? The Goodwe Support told me, they can do this remotely, but they never did.
Maybe this will solve my issues.

smileyman-1, 2024-06-10:
There are three updates for the inverter: ARM, DSP, communication module (WLAN 1.0, WLAN 2.0 or LAN)
ARM and DSP is the firmware of the inverter itself. Support can update this remotely; but only if the Home Assistant does not regularly query data internally via Modbus.
You can also update this yourself. However, a Goodwe Bluetooth stick is required for this with the EM series. - Do you have one?
You can find the ARM and DSP firmware here: https://drive.google.com/drive/folders/110hBxoNVC4M9F_GK5sKP7uq_aDNxwstZ
This can be installed via SolarGo if you have connected the Bluetooth stick instead of the WLAN or LAN module (reboot necessary to change). In the best case, SolarGo will also find the firmware itself. (SolarGo sometimes shows after the update that there is a new firmware and wants to install the same one again and again - just ignore that.)
There are also instructions from Goodwe for this whole procedure. If necessary, I can find them for you.
The aim is to have the ARM and DSP up to date first.
The communication module should also be updated to the latest version. Unfortunately, I only know the WLAN module 1.0, which is supplied as standard. I have the latest firmware for that. This is usually provided by the support. Unfortunately, I don't know it for the LAN module.
The WLAN module is originally manufactured by hi-flying.com. If necessary, there are also current versions here (which are unfortunately not compatible with SolarGo).

kolossboss, 2024-06-10:
I do not have an usb dongle.

I'm from germany (plz 212xx  30km südlich von Hamburg)
Maybe you can borrow me the stick?

smileyman-1, 2024-06-10:
Yes, we should find a solution.
Are you in the Discord group? Under what name?
Or how can I contact you?

kolossboss, 2024-06-10:
Yes,
I have discord.
My username ist SloppolS#2191

Thanks a lot.

dl6dbh, 2024-06-12:
give ist an disscord server platform for the integration ?
im run 9.9.25, quit good

JaniSchlumpf, 2024-06-13:
Just to add that version 0.9.9.27 works fine for me and I am not seeing the issues mentioned above. I recently switched from the official HA version to experimental as I was not able to set Eco mode. Everything works really well with 0.9.9.27 for me. As I am also having an EM inverter justed wanted to share my setup:

GW5048-EM
Firmware: 1314L / 410-04000-13 (seems the latest DSP from the mentioned GoogleDrive)
Goodwe WiFi / Lan Kit 1.0 (with USB - got this exchanged by GoodWe because the original WiFi Stick failed)
Connected via ethernet to my HA running on VM. 

Would strongly suggest to apply for a firmware update with GoodWe support. They can be a little reluctant I learned myself, just keep asking :-) . 
Regarding bluetooth dongle - got mine from Kleinanzeigen.de , always worth a try.

mletenay, 2024-06-15:
... just to let you know, I'm still trying to figure out what the heck is going on there.
Fortunately, one nice gentleman from Czechia send me one Wifi kit and one LAN kit to play with.
The Wifi kit (fw 1.1.2.1) works like a charm but the LAN kit (fw 1.0.14) is unreliable as hell and I get tons of truncated UDP packets like you do.
Which is actually a good news, since I can finally investigate it locally and it increases a chance I will be able to explain and potentially fix it.

kolossboss, 2024-06-16:
Hi, thanks for all the help.

I also get the Bluetooth USB Dongle from a very kind guy here Germany (thanks @smileyman-1 ).

I tried every to update to the latest firmware. But in SolarGo, the update menu was not displayed, and in PVMaster, I couldn't select the firmware or it wasn't recognized when I placed it in the share folder. I tried it with both iOS and Android.

I contacted GoodWe support, and they opened a ticket to upgrade the inverter's firmware. However, I have not received a response so far.

@smileyman-1 also sent me a WiFi stick with firmware version 1.1.2.1. Today, I installed this stick, and it's working great so far! No issues with v0.9.9.28, and I'm getting new sensor updates every 10 seconds inHA. I've never had such a stable connection before.

So, my LAN stick is definitely also an issue.

Now I'm waiting for the inverter update, or maybe I'll find a way to upgrade the LAN stick with a better working firmware.

JaniSchlumpf, 2024-06-16:
Just for the record: my "WiFi / Lan Kit 1.0" is on firmware V1.2.1.19 (Boot:V1.1.4) . No idea if this is the latest version, anyway it runs just fine with my setup.

mletenay, 2024-06-16:
So, I have tested it a lot today and came to these conclusions:

1) The troubles seem to be caused by the Wifi or LAN dongle, inverter firmware and network infrastructure are most likely not the problem. 
2) LAN Box (5-pin, E20-GDW-0001 P/N: 151-30054-00), fw 1.0.14 is buggy as hell and close to unusable.
3) WiFi Box (5-pin, fw 1.1.1.2) and Wifi Kit (USB, fw 1.1.2.1) work perfectly fine with 5s polling and SEMS still getting updates.
4) The LAN box is sending randomly truncated responses. The truncation happens somewhere inside the modbus reading logic, since the IP/UDP layer sends proper packets with valid length and checksum, the payload being sent is already truncated.
5) The LAN box seems to be failing more often when the same socket (UDP source port) is being re-used. (That might be the difference to 0.9.9.17). When socket is always closed, the chance the inverter will send proper response increases.
6) I have adjusted the code, so the "keep-alive" is now OFF by default to both UDP and TCP. I have also adjusted the code to close the socket even between the re-tries to increase the chance the inverter/dongle will send the proper request.
7) I have released the library and updated HACS master to contain these changes, it can be tested if it helps.
8) There is fw 1.0.15 for the LAN box dongle out there, I will try to get it somehow, to see if it is any better than the 1.0.14 I have now.

smileyman-1, 2024-06-16:
I can confirm most of the points.
The only thing that looks different for me is point 3: The WiFi Kit 1.0 can be queried every second by HA without any problems, but complete data is only delivered to SEMS at an interval of > 180 seconds.

janchlebek, 2024-06-17:
> I can confirm most of the points. The only thing that looks different for me is point 3: The WiFi Kit 1.0 can be queried every second by HA without any problems, but complete data is only delivered to SEMS at an interval of > 180 seconds.

But that is OK in my opinion - update frequency of SEMS depends on when the inverter decides to send data and is configured by Goodwe. It has nothing to do with HA polling.

janchlebek, 2024-06-17:
> @janchlebek SEMS update frequency is controlled by register 47006 (default 12, *5s = 1m) and no, for some users out there, when HA polls too frequently, SEMS somehow stops receiving updates. 

Ah, did not know this is configurable. Thanks for clarifying.

[1 more reply]