mletenay/home-assistant-goodwe-inverter #281: Entities today will stop working after midnight

mletenay, 2024-05-10:
### Discussed in https://github.com/mletenay/home-assistant-goodwe-inverter/discussions/279

<div type='discussions-op-text'>

<sup>Originally posted by **sitvesep** May 10, 2024</sup>
The last two updates are causing me to get an error as soon as it is midnight, when the entity states are supposed to be reset: **the entity is not numeric**.  Only entities **today** will stop working.  If I revert to  v0.9.9.17, everything is fine.

[image]
[image]

This problem I was also noticed by some people on the FB group about PV. 

Does anyone also have experience with this error ?

GW10K-ET
Firmaware: 04029-11-S11/02041-27-S00
 

Thank you

</div>

mletenay, 2024-05-13:
Well, I liked the 0 value too, but strictly technically speaking, that is not what inverter reports.
Inverter says N/A (0xFFFF/-1).
The hacky conversion to 0 was replaced in the library with None to consolidate behavior of Total and Daily sensors.
And the Total has to return None, otherwise HA's energy dashboard goes completely nuts ...
I need to find a proper way/place where to put None->0 for daily sensors only, which I do not like too much ...

Legnava89, 2024-05-13:
Unfortunately, I can't upgrade to a higher version until the issue is
resolved somehow.
Why does it work without problems in v0.9.9.17 and not in the new version?
The inverter keeps the same status.

mletenay, 2024-05-13:
Yes, the inverter behaves the same - returns N/A.
But the python library code was consolidated/fixed and now the N/A is propagated to HA.
As said above, the fact it worked before was kinda hack.

Legnava89, 2024-05-13:
If it can't be returned, I have a workaround via node-red that updates the status from unknown to 0.
But you have to put python_scripts in HA.
I'll try after midnight tonight. In HA development, this works through node-red.

mletenay, 2024-05-14:
Errrrrr, actually I was wrong above.
The inverter reports 0 not N/A.
I just made a mistake when fixing the DT behavior ...
Fix is on the way ...

Legnava89, 2024-05-14:
Thank you very much. I will praise you everywhere

sitvesep, 2024-05-14:
thanks for checking the error and fixing it. I believe it will help a lot of people. thanks for your work, coffee is on the way.

mletenay, 2024-05-16:
Fixed in v0.9.9.20

closed 2024-05-16 by mletenay

Legnava89, 2024-05-16:
After the new update, the complete cell integration does not work.
[image]
[image]
[image]

sitvesep, 2024-05-16:
yes, the integration has completely stopped working for me as well. downloaded the version back and it's ok

[home-assistant_goodwe_2024-05-16T22-26-37.131Z.log](https://github.com/mletenay/home-assistant-goodwe-inverter/files/15341637/home-assistant_goodwe_2024-05-16T22-26-37.131Z.log)
[image]

sitvesep, 2024-05-16:
I tried to delete the device and add it again and it works

mletenay, 2024-05-17:
Yeah, I forgot the need for settings migration, v0.9.9.20 was broken in this aspect.
v0.9.9.22 is fixed and works without need to re-add integration.