mletenay/home-assistant-goodwe-inverter #289: Flexible IP Address

jck5000, 2024-06-06:
Unfortunately, my single avaliable Internet provider only offers a s#!t router, which doesn't allow enough fixed IP adresses to be set by DHCP, and the Goodwe SolarGo app essentially removed the feature to just set a fixed IP (or at least it isn't working).

Yes, I've complained to them already. 

This (great!; thank you so much!) integration, though, asks for an IP adress. Which is _not_ re-configureable in the GUI, I essentially uninstall and reinstall the integration every time the s#!t router f%cks up.

Someone made an IP Scanner plugin on HACS, that works perfectly fine _and_ someone has already done that, so the only thing I need to do is open that, reat the title line which I typed in as "Goodwe inverter MAC is "...A2:D2", look up whatever IP is current, and then tell that to the Goodwe integration.

Now, my idea is, that if anyone knew how to do that, it should obviosly possible to find the inverter in a network if the MAC address is known, so this auto-updates.

Unfortunately, I have absolutely zero clue on how any of that works; I can't even guess the actual difficulty of doing that. ButI'd really appreciate this if someone could and would implement it.

janchlebek, 2024-06-06:
I honestly think this is out of scope for the integration. Dealing with a dynamic IP address is a network task and goes beyond what an integration should deal with. 
If you have such a problem with the stupid router why not put anorher one behind it and manage the network independently?

smileyman-1, 2024-06-10:
An idea for a better workaround: you can configure a fixed IP address in the web interface of the WLAN / LAN interface of the Goodwe inverter (open http://IP address once from a browser).

mletenay, 2024-06-10:
As mentioned above, solving network layer issues is not and will not be responsibility of the integration.
I believe the 2 suggested solutions could help you:
- it should be possible to set fixed IP via web ui of the inverter/wifi module
- or buying cheap router and using that behind your ISP router which will give you lot of other opportunities

closed 2024-06-10 by mletenay