mletenay/home-assistant-goodwe-inverter #291: One battery won't discharge from 100%

devappsoftware, 2024-06-11:
hi there, since installing the experiment version of the software and using the fast charge, one of the batteries stays at 100% and doesn't charge. There are 3 batteries, and 1 is pair together.. so the system sees 2 batteries. One discharges fine, but the other keeps at 100%. I've used the fast charge toggle button on a timer.

LX F16.4-H x 3 stacks.
Invertor: GW25K-ET (AS4777-2 2020)

I've tried restarting the whole stack including batteries with no end. I've depleted the batteries and the 100% charge one simply won't discharge. Is there a setting to check for this?

I do see this in the logs from time to time:
Diag Status changed to BMS: Discharge current low, APP: Discharge current too low, SOC delta too volatile, PF value set, Real power limit set.

Charge batteries: Error executing script. Invalid data for call_service at pos 1: expected str for dictionary value @ data['value']

Could it be the above values are not set properly? If so, what should the value be?
"yes", "no", null or something else?

Adam

adevapp, 2024-06-11:
Also found this in the logs:
Logger: homeassistant.helpers.entity
Source: helpers/entity.py:1176
First occurred: June 9, 2024 at 8:54:56 AM (5 occurrences)
Last logged: 1:34:36 PM

Updating state for sensor.pv8_voltage (<class 'custom_components.goodwe.sensor.InverterSensor'>) took 0.418 seconds. Please create a bug report at https://github.com/mletenay/home-assistant-goodwe-inverter/issues
Updating state for sensor.pv12_current (<class 'custom_components.goodwe.sensor.InverterSensor'>) took 0.409 seconds. Please create a bug report at https://github.com/mletenay/home-assistant-goodwe-inverter/issues
Updating state for sensor.battery_warning (<class 'custom_components.goodwe.sensor.InverterSensor'>) took 0.592 seconds. Please create a bug report at https://github.com/mletenay/home-assistant-goodwe-inverter/issues
Updating state for sensor.pv_channel (<class 'custom_components.goodwe.sensor.InverterSensor'>) took 0.464 seconds. Please create a bug report at https://github.com/mletenay/home-assistant-goodwe-inverter/issues
Updating state for sensor.today_battery_charge (<class 'custom_components.goodwe.sensor.InverterSensor'>) took 0.432 seconds. Please create a bug report at https://github.com/mletenay/home-assistant-goodwe-inverter/issues

adevapp, 2024-06-11:
Seems to be fixed by manually setting: battery_discharge_current comparing the 2 stacks to the same value.
Is there an officially recommended value for this? I'll check with Goodwe Support.

adevapp, 2024-06-11:
Seems 25 is the value based on this doc: https://www.goodwe.com.au/Ftp/Downloads/Datasheet/AU/GW_Lynx%20Home%20F%20PLUS+%20Series%20(HV)_Datasheet-AU.pdf

closed 2024-06-12 by adevapp