mletenay/home-assistant-goodwe-inverter #295: how to revert battery_charge_current to bms value?

bakerboy908, 2024-07-02:
I have setup an automation to limit charging current to 1A above 80% soc, this is working great, however the next morning when my battery is dead and should be charging at a faster rate it is still limited by the 1A set the day before.

Expected behavior
setting the charge current will be valid until the bms calls for a new charge current.

Current behavior
settings charge current override the bms called for value forever 

inverter: ET+ 10Kwh 
Battery: Goodwe Lynx F

nzkiwi68, 2024-07-03:
I use HA to drive my GoodWe GW5048D-ES and I also change the charge rate at night.
Batteries are Pylon Tech US3000 x 6 (~18 kWh total storage).

I have a complex calculation that sets the charger power level between 20% and 100% of the rated charger power (maxium is 4600 watts) based on my battery level for a charge between 9pm and midnight, this is called "Eco mode power" on the GW5048D-ES.

However, to your comment. I don't think the BMS "requests" as such a battery charge rate, but, rather states a maximum permissible charge rate. I think it's up to you to reset the charge rate.

I would simply set another automation to set the charge rate back up at sometime (say 4am) to whatever level you would like.

In my system, I actually flick the GoodWe from "General" mode to "Eco Charge" mode to start charging and then back to "General Mode" after charging so the charge rate off solar the next day is unrestricted. The GoodWe in General mode will charge my batteries at full speed regardless of the "Eco mode power" rate I have set. This is because that setting only takes effect if the GoodWe is "Eco charge" mode and not "General" mode.

I hope this is useful for you.

bakerboy908, 2024-07-04:
I wrote out this big message and then decided to check something.

My automation is being called more then i expected, its being called at 80% when the battery is charging and discharging so its being called again and resetting the value on the way down...

Silly me, ill close this issue now

closed 2024-07-04 by bakerboy908