mletenay/home-assistant-goodwe-inverter #299: Feature request: Negative value for Grid export limit

GIJOE-SA, 2024-07-29:
Hi, will it be possible to add the capability to make the export limit a negative value. 

On the new firmware for my GW6000ES20 inverter there is a constant 100W export from batteries when there is no PV Power and then a constant 100W import from the grid when there is PV Power. I have lodged the call with Goodwe and they have acknowledged as a bug but have not given any indication to when it will be fixed. The work around the have provided is to change the export limit to -100 through the SolarGo application. As it stand now I have to change it in the morning the limit to 0 when there is PV power and -100 if there is no PV power. I would like to automate this but the integration does not allow negative values. 

I have changed the minimum value for export limit number in the number.py to -100 but this alone does not seems to work. 

[image]

[image]

Thank you for the assistance.

rinners, 2024-08-21:
Does anyone know if there was ever a version of the firmware that didn't do this?

> On the new firmware for my GW6000ES20 inverter there is a constant 100W export from batteries when there is no PV Power and then a constant 100W import from the grid when there is PV Power. 

On the same model I get 100W export to grid when discharging batteries or when producing PV and 125W import to batteries if my DoD does not match up with my battery SoC. Unless I change the export limit to -100W in the apps

GIJOE-SA, 2024-08-21:
I know my problems with the batteries exporting to the grid started with version 07.7202.  However, going through my history, I suspect that it was still importing +- 100W constantly from the grid during battery discharge. Sorry I do not remember what version I had before 07.7202. 

Testing with 08.8808 now and early results seems like it works as one would expect if you set the export limit to -100W. 

Goodwe support is just not what it used to be as I am not getting any response from them on this issue.

rinners, 2024-08-29:
The new DSP firmware seem to be working for me too. Down to less than 0.1kWh exported per day with the export limit set to -100W in the app. The integration shows the the export limit selector at 10000 when hovering over on a Windows browser and 65,436W on mobile browser.