mletenay/home-assistant-goodwe-inverter #300: House Consumption and 2nd Battery

raa-hasel, 2024-07-30:
Now i have a 2nd Battery Lynx connected to my inverter GW25K-ET. For Battery 1 there is a sensor for "Battery Power" but not for the 2nd Battery. What i see now is, that the power of charge or discharge of battery 2 are counted/added to the House Consumption. 
This should be not the right way, or?
May be there is a Modbus-Register for "Battery 2 Power" ?

garv3, 2024-10-13:
It is exactly the same for me! Still no "Battery 2 Power". This makes the stats completely useless.

garv3, 2024-10-14:
@mletenay No commits since 3.5 months now. Is the project still being maintained?

mletenay, 2024-10-15:
Yes, it is. There just wasn't anything urgent to do recently, it just mostly works.
Regarding this 2nd battery, the situation is unchanged - we do not know where or whether that register exists.

raa-hasel, 2024-10-15:
Thanks for your update
Is house consumption not the calcution of the known power registers?
What is with the register 35264 (Battery2-Power)?

raa-hasel, 2024-10-16:
I found it on Github :-)
May be not offical, but it works for me
https://github.com/marcelblijleven/goodwe/blob/master/goodwe/et.py

garv3, 2024-10-16:
Okay, so now I understand what's going on. This repo uses the package `goodwe 0.4.8` from PyPI. In the et.py of this package some lines are commented out which include the battery 2 power.
```
        # Power4S("pbattery2", 35264, "Battery2 Power", Kind.BAT),
        # Integer("battery2_mode", 35266, "Battery2 Mode code", "", Kind.BAT),
        # Enum2("battery2_mode_label", 35184, BATTERY_MODES, "Battery2 Mode", Kind.BAT),
```
I will try to uncomment this line and check if the power is read correctly.

raa-hasel, 2024-10-16:
FYI:
I'm using the register via modbus (Waveshare) directly

garv3, 2024-10-16:
The script runs without errors but pbattery2 reports 0 W which is not correct:
`pbattery2:               Battery2 Power = 0 W`

@raa-hasel You wrote it works for you? Which inverter and ARM firmware are you using?
I have a GW25K-ET on ARM 08.401 (newer versions were buggy so we downgraded).

raa-hasel, 2024-10-16:
GW25K-ET / 2 x Lynx F 16,4 H / DSP 07.7068 / ARM 08.401 / BMS 6
See my last post (modbus)

garv3, 2024-10-16:
I am not sure how the PyPI goodwe package reads the data. I would have to dig into this.
But Maybe it is because my second BMS FW is 00 and won't show me an update. BMS1 FW is 07.
DSP and ARM are the same for our system. Hardware seems to be nearly identical too. With the difference that I am using 2x Lynx Home F Plus+ 13.1 batteries.

raa-hasel, 2024-10-16:
I do not have any problems with BMS 6, therefore ask Goodwe Support for theses Updates in advance.
SolarGoApp support only the update for BAT1 and till yet not for BAT2.

garv3, 2024-10-16:
> SolarGoApp support only the update for BAT1 and till yet not for BAT2.
This is a joke! WTH?
I will ask them to update the BMS. And report back if it did the trick. Thanks.

BTW: The PyPI package currently uses UDP because I currently have the old WiFi/LAN dongle installed. I will switch bach to the new 2.0 version soon and can test TCP then.

raa-hasel, 2024-10-16:
Be careful with 2.0 see German Forum if you can.
https://www.photovoltaikforum.com/thread/221719-goodwe-modbus-tcp-nur-%C3%BCber-wifi-lan-kit-2-0/

My opion is if BAT1-Power readable via UPD, BAT2-Power should it be too.

garv3, 2024-10-16:
Don't have the time to read the whole thread there right now. Can you give me a TLDR? I have used 2.0 over TCP before and everything seemed to work fine.

> My opinion is if BAT1-Power readable via UDP, BAT2-Power should it be too.

Yes, but maybe it is due to BMS FW 00... We will see as soon as the FW has been updated.

raa-hasel, 2024-10-16:
Sorry TLDR is ?

garv3, 2024-10-16:
"Too long, didn't read" - a summary. I am German, but reading the whole thread is just too time consuming for me now. So if you know what the conclusion of it is, I would appreciate a short summary why WiFi/LAN 2.0 should be avoided.

vikicv, 2024-10-28:
Hello, thanks for supporting the goodwe community!
In my configuration of GW25k-ET and 2x Pylontech batteries connected I am also missing variables related to 2nd battery.
Not bringing any news, just recap.
[image]

ccspack, 2025-06-10:
What's the chance that someone knows how long it is going to take to get the Battery 2 Power to show up in this Home Assistant HACs integration? 

As from what I've found the et.py file isn't accessible in HA, but I've had the Modbus address confirmed by Goodwe, to be 35265, on unit/slave 1.