mletenay/home-assistant-goodwe-inverter #302: MPPT Power and PV Power values mixed up for GW25K-ET

ChrisLuck, 2024-08-25:
Thanks for this amazing integration!
I spotted some mixed up values though. 

**Setup:**

Home Assistant OS 2024.8.2
Integration version 0.9.9.27
GW25K-ET with the WIFI/LAN KIT 2.0 over Wifi, all firmware up to date
East string connected to MPPT1, PV1
West string connected to MPPT2, PV3

**Problem:**

- `MPPT2 Power` is "unknown"
Only `MPPT1 Power` is correct.
I noticed that this was working correctly with Communication Module FW version 1.2.32. The power values disappeared immediately after the update to 1.3.53.
The values of MPPT1 Current and MPPT2 Current seem to be correct though.
[image]

- `PV2 Power` and `PV3 Power` are incorrect
`PV2 Power` is showing what should actually be `PV3 Power`.
The `PVx Current` and `PVx Voltage` values seem to 
be correct though. It is only the power which is mixed up.
[image]

---

This is what I got from the debugging logs:
~~~
2024-08-24 19:32:59.366 ERROR (MainThread) [homeassistant.components.sensor] Platform goodwe does not generate unique IDs. ID goodwe-meter_e_total_exp-9025KETT242L0036 already exists - ignoring sensor.meter_total_energy_export
2024-08-24 19:32:59.366 ERROR (MainThread) [homeassistant.components.sensor] Platform goodwe does not generate unique IDs. ID goodwe-meter_e_total_imp-9025KETT242L0036 already exists - ignoring sensor.meter_total_energy_import
~~~
~~~
2024-08-25 14:52:53.610 DEBUG (MainThread) [goodwe] Connecting to ET family inverter at 192.168.2.25:502.
2024-08-25 14:52:53.611 DEBUG (MainThread) [goodwe.protocol] Creating lock instance for current event loop.
2024-08-25 14:52:53.611 DEBUG (MainThread) [goodwe.protocol] Opening connection.
2024-08-25 14:52:53.624 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ZeroDivisionError: float division by zero') while processing template 'Template<template=({{ ((9700|float(0)) / (states("sensor.verdienst_pro_tag")|float(0))) | round(0) }}) renders=38>' for attribute '_attr_native_value' in entity 'sensor.amortisationsdauer'
2024-08-25 14:52:53.624 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: int got invalid input 'unavailable' when rendering template '{{ as_timestamp(now() + timedelta(days=(states('sensor.amortisationsdauer')|int))) | timestamp_custom('%b %Y') }}' but no default was specified') while processing template 'Template<template=({{ as_timestamp(now() + timedelta(days=(states('sensor.amortisationsdauer')|int))) | timestamp_custom('%b %Y') }}) renders=2358>' for attribute '_attr_native_value' in entity 'sensor.amortisationsdatum'
2024-08-25 14:52:53.680 DEBUG (MainThread) [goodwe.protocol] Connection opened.
2024-08-25 14:52:53.681 DEBUG (MainThread) [goodwe.protocol] Sending: READ 33 registers from 35000 (b52f00000006f70388b80021)
2024-08-25 14:52:53.832 DEBUG (MainThread) [goodwe.protocol] Received: b52f00000045f70342000061a80001393032354b4554543234324c3030333600000000000000000000000600061f6d0008019330343036322d30382d53303030323032302d30392d533031
2024-08-25 14:52:53.833 DEBUG (MainThread) [goodwe.protocol] Sending: READ 6 registers from 47547 (b53000000006f703b9bb0006)
2024-08-25 14:52:53.858 DEBUG (MainThread) [goodwe.protocol] Received: b5300000000ff7030c000000005500000000000000
2024-08-25 14:52:53.858 DEBUG (MainThread) [goodwe.protocol] Sending: READ 6 registers from 47589 (b53100000006f703b9e50006)
2024-08-25 14:52:53.931 DEBUG (MainThread) [goodwe.protocol] Received: b5310000000ff7030cffff00005500000000000000
2024-08-25 14:52:53.931 DEBUG (MainThread) [goodwe] Connected to inverter , S/N:9025KETT242L0036.
2024-08-25 14:52:53.932 DEBUG (MainThread) [goodwe.protocol] Sending: READ 125 registers from 35100 (b53200000006f703891c007d)
2024-08-25 14:52:53.960 DEBUG (MainThread) [goodwe.protocol] Received: b532000000fdf703fa1808190e343315c9000f0000033315c900000000022915a2000a0000000015a200000000000000000202090500141386000001b708f300141386000001a308d9001313860000019c0001000004f6fffffb3d000001a70000054509010005138600000000001d08e60005138600000000000808cc00041386000000000017000000e40000005c0000087a0000003c00000993000101d50000019f04011d590ead00000000000000000000000000020001000000000000000018820000005f000017e7000000e700590000000c00000000141c00f200000000000000000000000000050000000000000000000000000000020401c8000000000100
2024-08-25 14:52:53.961 DEBUG (MainThread) [goodwe.protocol] Opening connection.
2024-08-25 14:52:53.962 DEBUG (MainThread) [goodwe.protocol] Connection closed.
2024-08-25 14:52:53.965 DEBUG (MainThread) [goodwe.protocol] Connection opened.
2024-08-25 14:52:53.966 DEBUG (MainThread) [goodwe.protocol] Sending: READ 22 registers from 39000 (b53300000006f70398580016)
2024-08-25 14:52:54.038 DEBUG (MainThread) [goodwe.protocol] Received: b5330000002ff7032c00000000000000011000000000000005000001ff000000000000000000000000000000000000000000000000
2024-08-25 14:52:54.038 DEBUG (MainThread) [goodwe.protocol] Opening connection.
2024-08-25 14:52:54.038 DEBUG (MainThread) [goodwe.protocol] Connection closed.
2024-08-25 14:52:54.043 DEBUG (MainThread) [goodwe.protocol] Connection opened.
2024-08-25 14:52:54.043 DEBUG (MainThread) [goodwe.protocol] Sending: READ 125 registers from 36000 (b53400000006f7038ca0007d)
2024-08-25 14:52:54.140 DEBUG (MainThread) [goodwe.protocol] Received: b534000000fdf703fa00070000000a0000000100d10143f925fb3901af003a0047ff9dffd213860000000000000000000000d100000143fffff925fffffb390000010300000118ffffff95000001af0000016c000001c9fffff917fffff5e100020005000000000000000000000000ffffffffffffffffffffffffffffffffffffffffffffffffffffffff000000000003ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff0000000000003a370000000000003dfe0000000000001621000000000000817d0000000000001855000000000000109400000000000046f300000000000063031808
2024-08-25 14:52:54.141 DEBUG (MainThread) [goodwe.protocol] Opening connection.
2024-08-25 14:52:54.141 DEBUG (MainThread) [goodwe.protocol] Connection closed.
2024-08-25 14:52:54.145 DEBUG (MainThread) [goodwe.protocol] Connection opened.
2024-08-25 14:52:54.146 DEBUG (MainThread) [goodwe.protocol] Sending: READ 61 registers from 35301 (b53500000006f70389e5003d)
2024-08-25 14:52:54.242 DEBUG (MainThread) [goodwe.protocol] Received: b5350000007df7037a0000055c000200000000000000000000000000000000ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff000000000000ffffffffffff00000000ffffffff0333ffffffffffffffffffffffffffff000f000a000000000000000000000000000000000000000000000000000000000000
2024-08-25 14:52:54.243 DEBUG (MainThread) [custom_components.goodwe.coordinator] Finished fetching GoodWe data in 0.311 seconds (success: True)
~~~

garv3, 2024-09-25:
Same Error on my side. Happens since I switched to the new WiFi/LAN 2.0 adapter and TCP.

vikicv, 2024-10-28:
Hello, thanks for supporting the goodwe community!

I was playing with currently present variables in Mletenay's integration on my GW25k-ET to get values from connected PV1, PV3, PV5 strings of the solar plant. Please look at picture below. It is clearly seen what is mixed up and where to get the proper values.
[image]