mletenay/home-assistant-goodwe-inverter #309: POWER_VOLT_AMPERE_REACTIVE is a deprecated (Error from logs)

jellewie, 2024-09-16:
I got this error in my logs:

```
Logger: homeassistant.const
Source: helpers/deprecation.py:197
First occurred: 17:26:39 (1 occurrences)
Last logged: 17:26:39

POWER_VOLT_AMPERE_REACTIVE was used from goodwe, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'goodwe' custom integration
```

I have an
```
GW6.5K-ET
by GoodWe
Firmware: 04029-12-S11 / 02041-28-S00
Hardware: ETU 96500 
```

lordjson, 2024-09-17:
Same here.

Model GW15KT-DT

lordjson, 2024-10-06:
This was the cause of some kind of corruption of my installation.
I got the warning "this is a deprecated constant" from two different addons, disabling the first reduced my problems and it seems fully solved after also disabling this one.

dirkbe, 2024-10-16:
Got the message too but seems to be depreciated by 2025.9 and should be replaced by UnitOfReactivePower.VOLT_AMPERE_REACTIVE

Logger: homeassistant.const
Source: helpers/deprecation.py:197
First occurence: 09:45:55 (1 occurrences)
Last logged: 09:45:55
Zuletzt protokolliert: 09:45:55

POWER_VOLT_AMPERE_REACTIVE was used from goodwe, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'goodwe' custom integration

dirkbe, 2024-10-16:
Created pull request to fix #319

mletenay, 2025-07-03:
Done within v0.9.9.28

closed 2025-07-03 by mletenay