mletenay/home-assistant-goodwe-inverter #312: Scan Interval has no effect using TCP

garv3, 2024-09-25:
I am using the experimental version and recently switched to the new WiFi/LAN 2.0 Dongle and changed the protocol to TCP.
What I found is, that the "Scan interval" setting seems to be useless. It is set to 2 seconds but the sensors update every 20 seconds.
[image]
[image]
When I used UDP with the old module, everything worked fine when updating every other second.
Any suggestions?

crispe, 2024-10-18:
This is not an issue with the integration. It is the firmware on the WiFi/LAN module that will report the same values during 20 seconds. So even if the integration in scanning every couple of seconds it keeps will appear that it is not since no values change.

garv3, 2024-10-21:
Oh, wow. This is a joke. What is goodwe doing here?

closed 2024-10-21 by garv3