mletenay/home-assistant-goodwe-inverter #326: ILLEGAL DATA VALUE: Unable to set 'Fast charging SoC' value

zhouska, 2024-11-04:
I'm unable to set the target SoC value using the integration, I get the following error is I use the slider or an automation /script:

```
2024-11-04 09:59:37.356 DEBUG (MainThread) [goodwe.protocol] Sending: WRITE 42 to register 47546 (f706b9ba002a183a)
2024-11-04 09:59:37.397 DEBUG (MainThread) [goodwe.modbus] Response is command failure: ILLEGAL DATA VALUE.
2024-11-04 09:59:37.397 DEBUG (MainThread) [goodwe.protocol] Received exception response: aa55f78603e253
2024-11-04 09:59:37.397 DEBUG (MainThread) [goodwe.protocol] Socket closed.
2024-11-04 09:59:37.397 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139787611399808] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2761, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2804, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 996, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1068, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 122, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/goodwe/number.py", line 201, in async_set_native_value
    await self.entity_description.setter(self._inverter, int(value))
  File "/usr/local/lib/python3.12/site-packages/goodwe/et.py", line 669, in write_setting
    await self._write_setting(setting, value)
  File "/usr/local/lib/python3.12/site-packages/goodwe/et.py", line 685, in _write_setting
    await self._read_from_socket(self._write_command(setting.offset, value))
  File "/usr/local/lib/python3.12/site-packages/goodwe/inverter.py", line 122, in _read_from_socket
    result = await command.execute(self._protocol)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 453, in execute
    response_future = await protocol.send_request(self)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 166, in send_request
    await response_future
  File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 134, in datagram_received
    if self.command.validator(data):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/goodwe/protocol.py", line 613, in <lambda>
    lambda x: validate_modbus_rtu_response(x, cmd, offset, value),
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/goodwe/modbus.py", line 206, in validate_modbus_rtu_response
    raise RequestRejectedException(failure_code)
goodwe.exceptions.RequestRejectedException: ILLEGAL DATA VALUE
```

I run the following version of the integration ``0.9.9.27`` on a GW10K-ET configured this way:

[image]

This HA version:

```
Core 2024.10.4
Supervisor 2024.10.3
Operating System 13.2
Frontend 20241002.4
```

The only change I'm aware was a recent update to hew HA core version. 

Entities 'goodwe_fast_charging_power' and 'goodwe_fast_charging_switch' still work. I tried to restart the inverter and HA to no avail.

zhouska, 2024-11-04:
Just tested it using SolarGo application. Am able to start 'Immediate charging' without any issues after I set required SoC and power values...

zhouska, 2024-11-04:
ok, an unexpected plot twist. I captured traffic from SolarGo, then replayed it and was able to set the SoC value to 87, which got me think if I can do the same using goodwe pip module. Long story short, yes, I'm able to do that, but, the lowest possible value I can write to the register is 78. 

I'm on firmware version ``111127``. Can someone verify my findings? It would basically mean the slider can only go from 78 - 100 for fast_charging...

Unlik, 2024-11-11:
Similar issue, I am trying to set number.goodwe_grid_export_limit to some value e.g. 10000 (does not matter which one). The value is set properly but still I get error:
`HomeAssistantError: Validation error: Value 10000.0 for number.goodwe_fast_charging_power is outside valid range 0 - 100`
I suppose it is somehow related?

zhouska, 2024-11-18:
@mletenay can you have a look here https://github.com/mletenay/home-assistant-goodwe-inverter/issues/326 and https://github.com/mletenay/home-assistant-goodwe-inverter/issues/327 since you maintain https://github.com/marcelblijleven/goodwe and have committed to https://github.com/marcelblijleven/goodwe in the past, please? 

I'm really at my wit's end as I have tried every trick I had up my sleeve already, such as:

- enabled debug logging to observe error messages
- captured traffic from HA to the inverter
- captured traffic from SolarGo to the inverter
- tried to set multiple fast charging SoC values in both HA and SolarGo (with mixed results, HA consistently fails with values below 78, SolarGo occasionally succeeds with values lower than 78 - 55 for instance - but these never stick once you go above 78 and try to go back to a lower value)
- replayed packets from SolarGo to the inverter using ``goodwe`` methods (outside HA) and using ``goodwe`` methods (still outside HA) with some success, but having intermittent failures for values below 78

The only thing I can see changing some time ago was: HA version and upgrade of the extension (switched to fast charging quite some time ago when it become available and was consistently using SoC values as low as 40% with no issues).

[@Unlik  for visibility]

feutl, 2024-12-04:
Having the same issue - and just realised it as I wanted to start using fast charging instead of switching my inverter to eco mode over night and charge it with these settings.

feutl, 2024-12-05:
Goodwe Addon says
Firmware: 04029-11-S11 / 02041-27-S00
Hardware: ETU 96500

HomeAssistant is on 
Core 2024.11.3
Supervisor 2024.11.4
Operating System 13.2
Frontend 20241106.2

zhouska, 2024-12-05:
it is the same version as I have:

Firmware: 04029-11-S11 / 02041-27-S00

deveylder, 2025-01-17:
I am on the latest arm version 30.

I get ILLEGAL DATA VALUE if under 92.

Everything is Ok between 92 => 100

version 0.9.9.27

GW10K-ET
door GoodWe
Firmware: 04029-12-S11 / 02041-30-S00
Hardware: ETU 9010K

yutani42, 2025-01-20:
For me it seems all values below the current SoC (54%) results in the error (maybe a coincide).