mletenay/home-assistant-goodwe-inverter #327: Validation error for the value which is not set

Unlik, 2024-11-12:
Hello, I am running on 
```
Core 2024.11.1
Supervisor 2024.11.2
Operating System 13.2
Frontend 20241106.2
home-assistant-goodwe-inverter 0.9.9.27
```
with Goodwe Inverter ET series on firmware 1127. After upgrading from 0.9.9.13 to the latest 0.9.9.27 I am facing the issue:
**if wants to set number.goodwe_grid_export_limit to some value e.g. 10000 then the value is set properly but still I get error:**
`HomeAssistantError: Validation error: Value 10000.0 for number.goodwe_fast_charging_power is outside valid range 0 - 100`

Please note: I am automatizing in NODE-Red, so it means the error is preventing the automation from continuing on the node.

janchlebek, 2024-11-12:
You mention in your example you are trying to set "some value e.g. 10000". Clearly in the error you see the number expected is between 0 and 100.

janchlebek, 2024-11-12:
These values are normally set by percentage of inverter rated power so for example to set export limit to 2500W on 10K inverter you would enter 25. Give it a try. 😉

Unlik, 2024-11-12:
@janchlebek Of course, I tried it, e.g. 50. Then the **grid_export_limit** is set to 50W (in percentage it should be 5000). And still got the same error. 
So two things are on my mind:
-  the settings of **grid_export_limit** in W working properly - should stay in W but checking the range is wrong here, too late here (bc the value is properly set)  and the error message is wrong (copied from a different place)
- or settings/checking of **grid_export_limit** is somehow somewhere in the sources wrongly mixed with **goodwe_fast_charging_power**

zhouska, 2024-11-15:
I'd suggest to try the same I did, namely try to capture traffic to the inverter from SolarGo and see if you can set the export limit in the app. I think it will fail the same way. I'm suspecting it is a 'feature' of the 1127 firmware.

I was able to intermittently to set SoC values lower than 78%, like 55% (in my case), but it wasn't persistent change, more info here: https://github.com/mletenay/home-assistant-goodwe-inverter/issues/326.

Perhaps a support case with Goodwe support should be opened for the firmware in question, if I only could find the link...

Unlik, 2024-11-15:
Hello @zhouska, That's a good point. I have tried to set up the export limit directly in SolarGo, and no issue has occurred. I can properly set tested values 50 or 10000 there without any errors. Also, with prior version of this integration (v0.9.9.13) I had no error.
Therefore I still think it is a valid issue of mletenay/home-assistant-goodwe-inverter in the current version v.9.9.27

zhouska, 2024-11-15:
> Hello @zhouska, That's a good point. I have tried to set up the export limit directly in SolarGo, and no issue has occurred. I can properly set tested values 50 or 10000 there without any errors. [...]

What I meant is that you could do the capture in SolarGo and in HA and compare the output. Maybe even try to replay the captured packet from SolarGo using python goodwe extension, like I did.

It did work with mixed results in my case, as I could go lower than 78% in SolarGo (at times, but changes didn't stick, and once I went over 78% and tried to go back I got same error), but replayed packets or newly crafted ones outside the SolarGo weren't accepted by the inverter. There seems to be some 'validation magic' done at inverter's side which is most likely related to how often one polls it for data...