mletenay/home-assistant-goodwe-inverter #329: Import power limit and power shaving mode

tomaswallentinus, 2024-11-23:
Hi.

In [issue 259](https://github.com/mletenay/home-assistant-goodwe-inverter/issues/259) you mentioned that peak_shaving_power_limit and peak_shaving_soc had been implemented. But I can't find it in the control of the inverter.

The power limit could be useful to prevent high spikes. Here in Sweden we just got a new billing system where the three highest peaks are billed monthly. So If I in homeassistant can trigger peak_shaving_mode and peak_shaving_power_limit when power is above a set value I can keep the peaks down.

Regards Tomas

P18home, 2025-01-20:
Yes  when I Use peak_shaving_mode it hade been perfect, if I can change peak_shaving_power_limit
to have have 4 kW at night and 2 at daytime , Becuse a big gridowner har half price night time( only half effect counts).
Now I need to use SolarGo App and change manuelly 22 and 6 o'clock , no so fun
I send a question to GoodWe that I want to more then one schedule for peak shaving  ,but they just closed the errend

ABEC7, 2025-01-28:
> Hi.
>
> In [issue 259](https://github.com/mletenay/home-assistant-goodwe-inverter/issues/259) you mentioned that peak_shaving_power_limit and peak_shaving_soc had been implemented. But I can't find it [...]

Are you able to even set peak shaving mode theough HA? When i try to set peak_shaving_mode the inverter never fully switches to peak shaving. I need to also go into solargo and select a power shaving limit, only then peak shaving mode starts working.

P18home, 2025-01-28:
I also need to use SolarGoApp (I also found it in the  SEMS+Portal app now)  to set and get power shaving mode , im using HACS GoodWe Inverter (experimental)  on Goodwe version 0.9.9.27 I think that the latest version 
But I should be very happy if I can set  peak_shaving_power_limit and peak_shaving_soc using HACS/HA that was mention but I think it was to handle fast charge/decharge and that is not  pefect becuse it just handel how much charge /decharge to/from the battery and power shaving checks aginst grid load that is what I want to balance against to get low effect cost