mletenay/home-assistant-goodwe-inverter #335: Battery state hold

klecksi, 2024-12-15:
Could the battery state hold be implemented, which can be trigged on register 47511 injecting 0x08 and trigged back in normal mode by 0x01?

Sinus76, 2024-12-19:
Not sure about the registers you mentioned but I use the "Inverter Operation Mode" to switch to the different modes, like Backup (state hold) or General (normal mode). I have it set up in an automation to exactly follow my dynamic energy contract.

klecksi, 2024-12-19:
I'm referring  to the battery state, so if the battery is  on hold it will not charge or discharge in that mode. I want to make use of this as well for dynamic pricing, more specific to discharge the battery just on designated times.

DevineCZ, 2025-03-05:
I use to set the battery to standby in HA directly writing to the modbus register. I have made two scripts, one to set the battery to standby and one to return it to normal. It works very well. Here is the configuration:

Standby script:

data:
  hub: battery_standby
  unit: 1
  address: 47511
  value: 8
action: modbus.write_register

Normal script:

data:
  hub: battery_standby
  unit: 1
  address: 47511
  value: 1
action: modbus.write_register

I have the following code in configuration.yaml:

modbus:
  - name: "battery_standby"
    type: tcp
    host: IP address Goodwe inverter
    port: 502
    delay: 0
    timeout: 5

    sensors:
      - name: "Battery standby status"
        address: 47511
        input_type: holding # To read a value from the holding register
        scan_interval: 10
        slave: 247

input_boolean:
  battery_fve_control:
    name: "PV battery control"
    initial: off
    icon: mdi:battery

switch:
  - platform: template
    switches:
      battery_fve_control:
        friendly_name: "Switching PV battery to Standby"
        value_template:"{{ is_state('input_boolean.battery_fve_control', 'on') }}"
        turn_on:
          service: script.switch_battery_fve_to_standby # Execute script to switch to Standby
        turn_off:
          service: script.switch_battery_fve_to_online # Execute script to switch back to Online

In order to get the switch between these mods to work properly, I had to create two more automations to check the status of the switch.

entity_id: input_boolean.battery_fve_control
to: "off"
trigger: state

and

entity_id: input_boolean.battery_fve_control
to: "on"
trigger: state

I use in other automations when charging EVs and when controlling PV plants with spot prices.

But it would be great if the integration made this ems_power_mode directly switchable from the integration as inverter operating mode.