mletenay/home-assistant-goodwe-inverter #336: Setting Working Modes

thisIsLiving, 2024-12-21:
I have a 25kW ET inverter and seem to have issues syncing or perhaps setting the working modes of the inverter via Home Assistant addon. When I use HA to change the working mode to "Self Use", the Working Mode in SolarGo says "Self Use" so that is fine. However when I change to the following modes, SolarGo shows the following:

|HA Working Mode | SolarGo Home Page Working Mode | SolarGo Quick Settings Working Mode |
| ------------- | ------------- | ------------- |
| General | General Mode | Self Use |
| Off Grid | Off-grid | Peak shaving |
| Backup | Backup Mode | Peak shaving |
| Eco mode| TOU Mode  | Peak shaving |
| Peak shaving | Peak shaving  | Peak shaving |
| Self Use | Self Use | Peak shaving |
| Eco charge | TOU Mode  | Peak shaving |
| Eco Discharge | TOU Mode | Peak shaving |

Is this normal??? And when I change the modes via the SolarGo app v5.9.0, the mode doesn't get updated in HA. Below is a screenshot of the firmware versions I have. Is it also possible to get the Smart Charging mode made available in the HA addon? It looks like it's available in the PyPI library.

[image]

saltydog256, 2024-12-26:
I have a 10kW ET and I have similar problems.  Using the sliders or buttons to set anything has a random result. Same with a 5 kW SBP, 5 kW EM and a 3kW EM.  I have tried different polling rates and timeouts to no avail.  The "read only" data is fine and reliable. Maybe there is an order that the must be executed to mimic SolarGo?

uitgeslapen, 2025-01-24:
I seem to have the same issue, with a 10kw ET.

ABEC7, 2025-01-28:
Same here with 15 ET. Unable to do anything else than set immediate charging.

une18, 2025-02-03:
Same question also from me. I have Fast charge power slider and Fast Charge ON/OFF switch. Inverter is in Self use mode and when I click this it starts scharging. With these I could build all kind automations.

How to set the inverter to Discharge to grid? Would like to Not leave General mode, as I am controlling the inverter also with 3-rd party controller. Need to overwrite some commands, but can not find a way. 
Similar Switch Fast Discharge and Power slider would be perfect!