mletenay/home-assistant-goodwe-inverter #344: Peak Shaving" mode in the Goodwe integration

deveylder, 2025-02-05:
Hi all,

I can change the inverter mode to peakshaving/general in the goodwe app in HA . 
In sems+ You can see peak or general mode activated.

But it does not really activate. To activate it i need to go into sems+ and click on the save mode button. so that the mode actually starts working.

Anyway looking for a solution to change to peakshaving and general mode from within ha goodwe integration without the need to manually safe the selected mode in Sems+. What am i doing wrong?

Intention is when the battery hits ex.. 10% => activate peakshaving. when battery is back to 25% => activate general mode. The reason for this is when you leave peakshaving mode active the battery never goes into standby and is wearing out.

Additionally i don't understand the button synchronize clock. But it is not a solution to safe the parameter for the mode.

thank you

DevineCZ, 2025-03-05:
In order for the peak shaving mod to work correctly in the goodwe integration, the modbus register 47542 - peakshaving power limit must be set correctly. If it works using sems+, it would be useful to compare the state of registers 47511, 47512 and 47542 in the normal inverter mod and after switching to peak shaving mod in sems+.
You can check the status of the registers e.g. in solargo by holding for a few seconds on the firmware version.

deveylder, 2025-03-13:
Thank you for your response. all works but if you switch to any mode ex. self_use within HA and then back to peak_shaving. It sets the repeat function to unknown instead of daily. So peakshaving is activated but not completely active.

In sems you can manually reactivate to daily. This does not happen in HA

ivik, 2025-08-15:
> In order for the peak shaving mod to work correctly in the goodwe integration, the modbus register 47542 - peakshaving power limit must be set correctly. If it works using sems+, it would be useful [...]

Hi

Can you explain a bit with what you mean with the following: "You can check the status of the registers e.g. in solargo by holding for a few seconds on the firmware version."

Is that under settings -> device information?

oceaneyes86, 2025-08-31:
I did some debugging. Seems that it's register 47591 that needs to be set on 64639. When I changed from Self Use mode to Peak shaving mode in Home Assistant that value isn't changed. Manually modifying it through the app to the value 64639 seems to activate the peak shaving mode.

EnigmaTD, 2025-09-05:
@oceaneyes86  Do you only set the value once? Or need to set it every time after switching back to Peak Shaving after switching to another mode?