mletenay/home-assistant-goodwe-inverter #348: How to sell from battery?

une18, 2025-02-10:
For Charging I have "Goodwe Fast Charging Switch" and " Goodwe Fast Charging Power" slider. I can not find similar button and slider for discharge. How to activate sell command from battery?

deveylder, 2025-02-12:
chane mode to eco but problem is the app does not do it. You need to do it in sems+

une18, 2025-02-13:
I am using 3rd party control device that works over RS485 and General mode is mandatory.  I need to overwrite some commands. Eco mode is not an option.

DevineCZ, 2025-03-05:
I was solving the same problem and solved it using modbus.write_register 47511.
EMS power mod 0x000C can be used
At the same time, the value set in register 47512 determines how many watts will go from the battery to the grid - here the value is 0 to 10000

deveylder, 2025-03-13:
How did you write into the modus register. Are you using the RS485 ?

arkovda, 2025-07-01:
There are (better)other ways to do this.
Use HACS and "GoodWe solar inverter for Home Assistant (experimental)"
works for me

tomaswallentinus, 2025-07-01:
This is the experimental repo

GregOhh, 2025-07-20:
> I was solving the same problem and solved it using modbus.write_register 47511. EMS power mod 0x000C can be used At the same time, the value set in register 47512 determines how many watts will go from the battery to the grid - here the value is 0 to 10000

I have been working on the modbus registers to command battery charging from the grid using HA. I am watching how solar go app does it and  mimicing the process. I am getting close to success but I am also  interested in the next step which is to command a selling to grid event. Those reg numbers will be helpful. I can't find an app that actually does that yet to watch but it looks like eco mode might be worth a look. I am using goodwe et 15kw. Any thoughts or help appreciated.