mletenay/home-assistant-goodwe-inverter #353: Release plans

jwierzbo, 2025-03-10:
Hi!

Is there any roadmap for official release which will be integrated into HA?
I'm using HA version and would like to get `Fast Charging` button available.

Also is possible to switch from official HA to this HACK version without data loss?

Thanks for your work!

janchlebek, 2025-03-10:
Hi,
you can switch between HACS and the official one, I did it in the past. As long as entity names remain same it should keep all data and config.
However I strongly recommend doing a backup (or snapshot if you run virtual).

mletenay, 2025-07-01:
The project is not dead, there just was no immediate need for a release (I will produce one this week).
As for the integration to HA, this will not happen. This HACS version is intentional superset of features for more advanced users which might be a bit too dangerous for average users and/or features which are not compliant with HA core policies.
The HA and HACS versions are compatible and freely interchangeable, you can install/uninstall HACS without data loss any time,