mletenay/home-assistant-goodwe-inverter #357: peak_shaving_power_limit and peak_shaving_soc settings values.

P18home, 2025-03-26:
Peak_shaving_power_limit and Peak_shaving_soc settings values.
It should be fantastic if those values can be in next release
I see in earlier issues that it has be developed and will be deployed in next release ,but I can't found them yet.

oceaneyes86, 2025-08-21:
I agree. I've tried to fix it using a script as mentioned in another thread https://github.com/mletenay/home-assistant-goodwe-inverter/issues/363 but it doesn't seem to work.

sequence:
  - action: goodwe.set_parameter
    metadata: {}
    data:
      device_id: 319c7f883bcxxxxxxxxxxxxxxxxxx
      parameter: peak_shaving_power_limit
      value: 2
alias: Goodwe - Set peak_shaving_power_limit
description: ""

magnusgyl, 2025-09-05:
Did you manage to solve this? I make a script and fails at:  Message malformed: required key not provided @ data['sequence'][0]['domain']

YordiDR, 2025-10-20:
@oceaneyes86 Did you figure out where the peak shaving values are located on GW5000-ES-20?
I have the same inverter (on latest firmwares) and have the same behaviour as you.
The registers you mention are listed in the modbus documentation i got from goodwe after-sales. Maybe i should pick it up with them.

magnusgyl, 2025-10-20:
Personnally I gave up and wrote my own script using eco.mode power and make HA change the power output from the battery

YordiDR, 2026-01-11:
@magnusgyl I find the WIFI/LAN 2.0 stick overall rather crappy so i'm re-implementing this with ESPHOME and RS485 RTU/Modbus instead of using the wifi/LAN stick.
I was going over all registers since i need to implement them all manually and i happened to find the correct registers for peak shaving reserved SoC (register 47593, uint16, precision 1%) and the peak shaving power threshold (register 47592, uint16, precision 0.01kW).

I figured i'd share it here as it could help others.

sebergh1234, 2026-01-12:
@magnusgyl 
> Personnally I gave up and wrote my own script using eco.mode power and make HA change the power output from the battery

interested in how you solved to write to the eco mode power parameter?

thisIsLiving, 2026-02-07:
I used some AI help me decipher how the code works in the GoodWe Python library and the HA GoodWe integration. It turns out you can access registers when you prefix the register number with "modbus-". I've used this to expose the register 47592 Peak Power Purchase Limit to HA. Have a look at this issue here and the code I used.

marcelblijleven/goodwe#108

thisIsLiving, 2026-09-12:
I've updated my inverter firmware and GoodWe must have changed the register location. With the help of Claude the new register for "Peak Power Sales Limit" is 47837, tested and working.