mletenay/home-assistant-goodwe-inverter #359: sensor.today_battery_charge and sensor.total_battery_charge too low

earstreet, 2025-03-27:
I have compared the data for charging the battery. It is noticeable that the calculated parameter (blue) corresponds almost exactly to twice the value read from the inverter. If you consider that I have a 5.12 kWh battery storage that is currently being charged from 5% to 100% and has an SOH of 98%, the result is as follows: 5.12 * 0.95 * 0.98 = 4.76672 kWh. So I would say that I trust the blue "own calculated"curve more than the orange one.
Additional information: Today and the day before yesterday it was so sunny that the battery storage could be fully charged.

[image]

This is how I calculate the blue parameter:
sensor.battery_power is given out by the inverter:

[image]

templates.yaml:
```
  - name: "calc_pv_to_battery_power"
    unit_of_measurement: "W"
    device_class: power
    state_class: measurement
    state: >
      {% set battery_power = states('sensor.battery_power') | float %}
      {% if battery_power < 0 %}
        {{ battery_power | abs }}
      {% else %}
        0
      {% endif %}
```

sensors.yaml:
```
- platform: integration
  source: sensor.calc_pv_to_battery_power
  name: calc_pv_to_battery_power_sum
  unit_prefix: k
  round: 1
  method: left
```

utility-meters.yaml:
```
calc_pv_to_battery_power_daily:
  source: sensor.calc_pv_to_battery_power_sum
  cycle: daily
```

sensor.total_battery_charge also shows a battery charge of approx. 5 kWh over 3 days, which is only half of the approx. 10 kWh that I add together from the three charge values in the first diagram.

[image]

Device info:
GW6.5K-ET
Firmware: 04029-12-S11 / 02041-29-S00
Hardware: ETU 96500
Integration version: 0.9.9.27

earstreet, 2025-03-28:
I think it might have something to do with the integration and temporal resolution of signals. I noticed that the steps in the blue curve in the first diagram are half the length of the orange signal. A signal with half the sampling rate would also only generate half the values during integration. So, in principle, it fits into the picture.

bakerboy908, 2025-09-09:
I believe goodWe has changed how that value is stored in the inverter, i am having issues with my battery (Goodwe Lynx F), so they upgraded my firmware to Firmware: 04029-10-S11 / 02041-31-S00 and then that issue started. :(

My battery software version was also changed from 3 to 9