mletenay/home-assistant-goodwe-inverter #360: Future Plans for releasing experimental features in the official HA build

xxgmxx, 2025-03-31:
Hello, and thanks for the great work on this integration!

I see that this custom component provides several advanced features not yet available in the official Home Assistant integration, such as Eco charge/discharge modes, and additional inverter controls.

Are there any plans to incorporate these features into the official Home Assistant GoodWe integration in the near future?

janchlebek, 2025-04-01:
I am not sure about that but I recall from some previous discussion that it should stay like this so only more advanced users use advanced features. 
If you want to use the full feature set I suggest you switch to experimental, you can do it anytime.

mletenay, 2025-07-01:
As I just mentioned in https://github.com/mletenay/home-assistant-goodwe-inverter/issues/353
This is intentional state. HA core is safe for average users, HACS is for users with special needs who are assumed to understand what are doing.

closed 2025-07-01 by mletenay