mletenay/home-assistant-goodwe-inverter #363: Add battery current limit settings

waynistan, 2025-04-29:
Hi,

I'm using this integration for my GW15K-ET hybrid inverter coupled with a LX F9.8-H battery (Firmware versions: DSP 10.10049, ARM 12.431, BMS 07).
In SolarGo version 6.2.2 the allowed battery charge & discharge current limits are adjustable under (in german) "Erweiterte Einstellungen" -> "Batteriefunktionseinstellungen" -> "Parametereinstellungen" (in english it should be something like "Advanced settings" -> "Battery settings" -> "Parameter settings" or similar).

Is it possible to make these current limit settings visible and adjustable in the integration? That would allow me to let the battery charge during the peak sunshine hours instead of the morning hours, if my pv prediction tells me there will be enough energy available to do that. This way the charging happens when it is the most favorable for the grid, without the risk of missing out on a fully charged battery right before sunset on a cloudy day (unlike a static setting, which SolarGo also offers I think).

Policanek, 2025-05-23:
Hi, I would also really appreciate this feature :-) I use the "number.goodwe_eco_mode_power" but it seems, that it doesn't work correctly.
Thank you for your effort.

mboron, 2025-08-08:
I would appreciate this feature too.

My use case is to slow down the charging during mornings when PV-generated power is low.
It allows keeping small overflows to the grid and avoids the inverter from consuming a small amount of power from the grid during battery charging.

JoshC1994, 2025-10-22:
Is there any plan to expose these values allowing it to be written directly from the intergration? With caveats of course. 

Also is there any way to restore the set values to the originals?
I see that the battery_charge_limit changes in respect to SOC, if i force set this, how can you get it to get it to return to the one defined by the inverter itself?
 
[image]