mletenay/home-assistant-goodwe-inverter #366: Data from the WiFi/LAN Kit-20 LAN module (WLA0000-03-00P) ... entity error during TCP connection

DvorakMartin1, 2025-05-05:
With the new version of the LAN module in the WiFi/LAN kit 20 (WLA000-03-00P), which only allows TCP connections, the incorrect value sensor.battery_current is read and the results are also reflected in the entities sensor.total_battery_charge and sensor.total_battery_discharge.
This happens even after uploading the new current version fw 2.2.29

[image]

KG887, 2025-05-19:
[image]

hbpv, 2025-07-04:
> i have ET G2 with wifi lan kit 2, I have enabled Modbus TCP via the Goodwe IoT platform, but i cannot connect using this plugin in home assistant: I get immediately "Failed to connect" (even when i select TCP)

Have you activated Modbus in the SolarGo app? Just put it to "On" - no other changes needed:

[image]

KG887, 2025-07-04:
Hi, In SolarGo you just set up the RS485 communication. 
i can not remember in Detail, but i gues the default com adress of 245 was causing issues on my site. I put it to 247 and it works.