mletenay/home-assistant-goodwe-inverter #369: POWER_VOLT_AMPERE_REACTIVE was used from goodwe, this is a deprecated constant which will be removed in HA Core 2025.9

deveylder, 2025-06-01:
Logger: homeassistant.const
Bron: helpers/deprecation.py:222
Eerst voorgekomen: 16:27:40 (1 gebeurtenis)
Laatst gelogd: 16:27:40

POWER_VOLT_AMPERE_REACTIVE was used from goodwe, this is a deprecated constant which will be removed in HA Core 2025.9. Use UnitOfReactivePower.VOLT_AMPERE_REACTIVE instead, please report it to the author of the 'goodwe' custom integration

Dodo7308, 2025-06-03:
Hi,
I've got the same warning message.

I found out that in the *sensor.py* file, you need to make the following
change:
*1. *Remove import POWER_VOLT_AMPERE_REACTIVE from constants import
*2. *Add import UnitOfReactivePower from homeassistant.const
*3. *Change *native_unit_of_measurement=POWER_VOLT_AMPERE_REACTIVE* to
*native_unit_of_measurement=UnitOfReactivePower.VOLT_AMPERE_REACTIVE* in
the definition for "var" sensors

This is exactly what Home Assistant requires - switching from the
deprecated constant *POWER_VOLT_AMPERE_REACTIVE* to the new
*UnitOfReactivePower.VOLT_AMPERE_REACTIVE*.
Don't forget to restart Home Assistant for the changes to take effect.

After this change, the warning message in the logs should stop appearing.

Kind regards

Jozef

mletenay, 2025-07-03:
Done within v0.9.9.28

closed 2025-07-03 by mletenay