mletenay/home-assistant-goodwe-inverter #375: eco_mode_power doesn't limit the import power using newer firmware

prezervos, 2025-08-12:
After firmware upgrade (currently DSP: 12.197, ARM: 31.60334) eco_mode_power is not working. I have to limit the battery charge current otherwise it charges battery with the full power. Before the firmware upgrade it was working correctly. Even when eco_mode_power set to 1, it charges the battery with full power from the grid.

nzkiwi68, 2025-08-14:
I had a bunch of strange issues like that too. Here's what I did and seemed to get resolved:

- Reboot HA completely, not a reload, but a reboot
- Disabled the GoodWe HACS and then renable it

After that, it all worked fine again.

prezervos, 2025-08-15:
Helped, thank you.

> I had a bunch of strange issues like that too. Here's what I did and seemed to get resolved:
> 
> * Reboot HA completely, not a reload, but a reboot
> * Disabled the GoodWe HACS and then renable it
> 
> After that, it all worked fine again.