mletenay/home-assistant-goodwe-inverter #376: goodwe_inverter_operation_mode erroring

tux43, 2025-08-15:
When I run this

action: select.select_option
metadata: {}
data:
  option: eco_discharge
target:
  entity_id: select.goodwe_inverter_operation_mode

It throws this in home_assistant.log

  File "/usr/src/homeassistant/homeassistant/components/select/init.py", line 188, in async_handle_select_option
    await self.async_select_option(option)
  File "/config/custom_components/goodwe/select.py", line 136, in async_select_option
    await self._inverter.set_operation_mode(
        _OPTION_TO_MODE[option], self._eco_mode_power, self._eco_mode_soc
    )
  File "/usr/local/lib/python3.13/site-packages/goodwe/et.py", line 764, in set_operation_mode
    raise ValueError()
ValueError

jkairys, 2025-12-19:
I see this too, other modes seem to be ok

Rene-Sackers, 2026-02-08:
@jkairys I posted a pretty long response here on what worked for me: https://github.com/mletenay/home-assistant-goodwe-inverter/issues/382#issuecomment-3868012758

Thanks @vt-code8 , your comment was one of the key parts to getting this fixed for me!

Rene-Sackers, 2026-02-15:
@jolinda2 you need the new (yet unreleased) version. I believe this is the way: https://github.com/mletenay/home-assistant-goodwe-inverter/discussions/395#discussioncomment-15795258

But don't quote me on it, haven't done it yet, hope to this weekend.